rosflight/rosflight_firmware #7: Mavlink issues

dpkoch, 2016-05-23:
Mavlink is having a couple of issues:

1. Sometimes doesn't start streaming on startup; have to send stream rate commands to get it started
2. Parameters don't play nice with QGroundControl anymore; it always misses one or two of them when requesting the parameter list

superjax, 2016-06-01:
Is this issue still open?

dpkoch, 2016-06-01:
I think it's partially resolved. I think the streaming on startup is fixed; as you've been using it do you consistently get a heartbeat and data streams?

Last time I tried, the param interface was still not playing nice with QGroundControl. The fcu_io2 node receives all of the parameters, but QGroundControl always misses one or two (not necessarily the same ones) and seems unable to recover. Still not sure what's causing that.

superjax, 2016-06-01:
Yeah, I haven't had any problem steaming IMU.  In fact, the other day when i was playing with my ground station, I had it streaming at 500 hz with no problems!  Asking it to do 600 seemed to cause problems though.  Haha.  

I wonder why qgroundcontrol is missing params.  Could it be that we are just sending the initial batch of params too fast?

dpkoch, 2016-06-02:
So it turns out the param_request_read message can ask either by name or index, and I had only implemented by name. QGroundControl and mavros use the index method to request params it missed from the initial list, which was why they couldn't recover. They recover now.

However, I'm still don't know why it misses some params in the first place; it makes me worry that I'm smashing messages together or something. So I'm going to leave this issue open until I investigate that.

superjax, 2016-06-02:
Sounds good!

superjax, 2016-06-16:
Does this have anything to do with BYU-MAGICC/fcu_io2/#9

closed 2016-06-17 by superjax