rosflight/rosflight_firmware #11: IMU Mavlink

superjax, 2016-05-27:
Hey @dpkoch,

I've been thinking a lot about trying to get good IMU timestamps.  I talked with Dr. McLain, and he pointed out that we can often make do with just knowing the difference between measurements, rather than the absolute timestamp.  I really liked that idea, and I'm wondering if you could add a field to the IMU message (could be just int16), for a time stamp.  It would be defined as the number of microseconds since the last IMU measurement was taken that this measurement was taken.

Simon pulled my changes to Breezy so we have an interrupt routine to play with to get the IMU timing data, so it should be pretty straight-forward.

dpkoch, 2016-05-30:
So I'm just trying to think how exactly we'll use the timestamp on the ROS side. The sensor_msgs/Imu message type has a header with a timestamp, but how will we populate that timestamp from a dt? Would we lose or degrade that info by using an Imu message?

superjax, 2016-05-30:
I don't think so.  in theory, the computer's clock and the naze's clock shouldn't wander off from each other significantly.  However, there is no guarantee that they are synced up.

1 - We could publish a /clock server out of fcu_io2 to sync up the whole system with the naze
2 - We could use ideas from the PTP (precision time protocol) process to sync up the naze's time with the computer
3 - We could (for now) just offset the IMU timestamp on the ROS message starting from the first IMU message.  I think we could see how much the timestamp on the message deviates from the computer time as we go longer and longer.

What do you think?

dpkoch, 2016-05-30:
Another option might be to still send the actual timestamp (microseconds since boot) over mavlink instead of dt and put it straight into the sensor_msgs/Imu header with an offset (either time since the first IMU message was received or something more sophisticated like the PTP stuff). If we end up wanting to use dt instead, we can just do the subtraction between timestamps on the ROS side.

superjax, 2016-05-30:
That's a good idea.  I am concerned with losing accuracy due to variability in the serial transfer, so using the time since boot would be just as accurate.  Let's do that.

superjax, 2016-06-06:
Is there anything we need to do in order to close this issue?

nonrevlb, 2016-06-14:
We put the baud_rate up to 921600, then increased the value of STRM_IMU.  The value of STRM_IMU doesn't line up with the actual average rate.  The average rate tends to be below the parameter by about 50 up to STRM_IMU=~800.  Setting STRM_IMU between ~800 and ~1100 always gives an average rate around 725.  If you go above 1100, the average rate jumps to ~1080.

superjax, 2016-06-16:
Thanks Brandon, I'll try to look into this and see if I can get an idea what is causing this behavior

dpkoch, 2016-06-16:
I've been needing to look into MAVLink timing stuff in general for a while. I opened issue #22; I can look into this if you'd like.

superjax, 2016-06-29:
I just added a timestamp to the SMALL_IMU message.  I used your script, but I didn't realize that it would just go ahead and push the master.  Sorry about that.  We are starting to put together the camera driver for the Naze, so we need the timestamp working.  We are thinking that we'll have the timestamp be the difference in us from the last IMU message per @dpkoch's suggestion.  I still need to get it working on the ROSflight2 and fcu_io2 side of things.

dpkoch, 2016-06-29:
So I've been thinking about this issue and I think we can do the time synchronization stuff to convert the absolute fcu reported time (ms or us since boot) to ROS time. My thought is that this is the way we'll want to do it in the long run because we need good absolute timestamps for our RMEKF implementation. Mavlink already has a time synchronization message type that we can leverage. I'll work on it this evening and let you know how it goes. Does that approach sound alright?

That's okay with the mavlink stuff. The mavlink_c_library repo is different; I see it more as just a place to dump stuff in an organized, versioned fashion more than a traditional git repo. It exists so that we can grab a specific version of the compiled headers as a submodule. That doesn't mean we should just push a bunch of garbage to it, but if there are some extra commits in there we don't ever actually end up using as a submodule then it's not a big deal. The important history happens in the mavlink submodule. So we should go through the pull request channel on the mavlink repo, but probably not so much on the mavlink_c_library repo.

superjax, 2016-06-29:
That sounds great.  Thanks.  I'll hold off until you get that working.

dpkoch, 2016-06-29:
Huh, I just realized that when you ran the script it pushed to the wrong repository... I'm kind of confused. The script is supposed to take your local copy of the `mavlink` repo, generate the C headers, then push the generated stuff to the `mavlink_c_library` repo. It looks like it somehow pushed your `mavlink` changes to master using the commit message it usually uses for the generated `mavlink_c_library` stuff... very bizarre. Any idea what happened? Did you get everything set up in your local `mavlink` repo according to the instructions at the top of update_c_library.sh (substituting BYU-MAGICC for mavlink in the URLs and mavlink_c_library for c_library)?

superjax, 2016-06-29:
Apparently I must have done it wrong.  I know that something was wrong.  I would have no problem with just blasting away my commits, and reverting back to where it was a while back.

PR #34 (2016-06-30): Time sync
Implemented time synchronization modeled after mavros and the PX4.

Time synchronization currently works for message types that include a time stamp. It would also be possible to estimate the delay and subtract that off of the time other, non time-stamped messages are received to make the resulting ROS time stamp a little more accurate, but I have not done that yet.

Closes #11.
closed 2016-06-30 by superjax