rosflight/rosflight_firmware #13: Muxing RC and offboard control

superjax, 2016-05-30:
Hey @dpkoch.  I'm working on the muxing of RC and offboard control.  I have written rc.c and I can fill the following data member

```c
typedef enum
{
  RC_RATE, // Channel is is in rate mode (mrad/s)
  RC_ANGLE, // Channel command is in angle mode (mrad)
  RC_THROTTLE, // Channel is direcly controlling throttle max/1000
  RC_ALTITUDE, // Channel is commanding a specified altitude
  RC_MIN_THROTTLE, // Channel will be compared with the offboard control input and the min throttle command will be taken
  RC_MIN_THROTTLE_ALTITUDE, // Channel will be the min throttle between command to maintain commanded altitude, and offboard throttle command
  RC_PASSTHROUGH, // Channel directly passes PWM input to the mixer
  RC_OFFBOARD // Channel is not being controlled by RC, but instead by offboard computer
} rc_control_type_t;

typedef struct
{
  rc_control_type_t x_type;
  rc_control_type_t y_type;
  rc_control_type_t z_type;
  rc_control_type_t F_type;
  int16_t x;
  int16_t y;
  int16_t z;
  int16_t F;
} rc_control_t;

extern rc_control_t _rc_commands;

```

This is in rc.h of the override branch.

Anyway, it looks like all the data fields are getting filled correctly.  I just need to get the mavlink stuff in so I can do the logic of the muxing.  Is there any chance you could help me get the mavlink into a similar state?  That is, a global variable with the four control channels, and a flag for each channel of it's type?  I know that the MAVlink message right now is a little different, but I'm actually not sure how to access it in the ROSflight code.  Is there a callback for offboard commands?  I can massage it into the right data fields if I know where to get ahold of it.

dpkoch, 2016-05-30:
I assume you're talking about the regular offboard control message, not the extended one?

So my understanding was that the mavlink message would have 1) an overall mode field (attitude, rates, altitude, passthrough) and 2) a bit field that just said whether it was sending valid commands for each channel or not. Then all of the logic you're working on here would happen onboard, but the user on the ROS side doesn't need to know anything about it. Is that still correct?

I'm going to try to draw up my understanding of how this was going to work, but won't we use the same enum and struct for the RC, offboard, and the combined command that goes into the control loops rather than have different one for each type (so `rc_control_type_t` would just become `control_type_t`, etc.)? My understanding was that we'd construct a `control_t` struct for RC and a `control_t` struct for offboard, then combine them as a function of the override mode, override switch, and stick deadbands to create a combined `control_t` struct that goes into the control flow structure and determines which loops get run for each channel. Are we thinking the same thing here?

superjax, 2016-05-30:
Yes, I'm talking about the primary 4 channels.

That's right, we wanted to have them all have the same type.  Should we create a mux.c which combines them?

superjax, 2016-05-30:
I think we also have a mode field for each channel.  Is that right?

dpkoch, 2016-05-30:
Yeah, a mux.c might be good.

So I thought that the mode would be an overall mode for the whole message. So you can do attitude control, or rate control, but we weren't going to support a mixed mode (e.g. roll angle and pitch rate) over the mavlink message. It seems to me that that would bloat the message (3 more bytes, which isn't necessarily a huge deal), and then you'd have to initialize all those flags every time when 99.9% of the time you'd never need the added flexibility. It seems to me like the added awkwardness and potential for error isn't worth a feature that would rarely if ever get used. What do you think?

superjax, 2016-05-30:
I agree with making a single field for the MAVlink message, however I think we may want the ability to mux RC rates with MAVlink angles.  So, within the ROSflight code, what if we had the additional byte field.  Here is what I am thinking for the combined message

```c
#include <stdbool.h>
#include <stdint.h>

typedef enum
{
  RATE, // Channel is is in rate mode (mrad/s)
  ANGLE, // Channel command is in angle mode (mrad)
  THROTTLE, // Channel is direcly controlling throttle max/1000
  ALTITUDE, // Channel is commanding a specified altitude
  PASSTHROUGH, // Channel directly passes PWM input to the mixer
} control_type_t;

typedef struct
{
  bool active; // Whether or not the channel is active
  control_type_t type;  // What type the channel is
  int16_t value; // The value of the channel
} control_channel_t;

typedef struct
{
  control_channel_t x;
  control_channel_t y;
  control_channel_t z;
  control_channel_t F;
} control_t;

extern control_t _rc_control;
extern control_t _offboard_control;
extern control_t _combined_control;
```

dpkoch, 2016-05-30:
Yes, I agree. Have just the main mode field in the MAVlink message but be able to mix different RC/MAVlink modes.

Looks great, that's right in line with what I was thinking. What is the `active` field for? What's the use case where a channel would not be active? Does it have to do with the extended passthrough stuff?

dpkoch, 2016-05-30:
Also, while I'm modifying the OFFBOARD_CONTROL mavlink message, let's make sure we're good with the control modes. Currently there's:

* `PASS_THROUGH`
* `ROLL_PITCH_YAWRATE_THROTTLE`
* `ROLL_PITCH_YAWRATE_ALTITUDE`

We'll need a rates mode, perhaps:

* `ROLLRATE_PITCHRATE_YAWRATE_THROTTLE`

although that name's getting pretty long. Do we want modes where we control yaw directly? E.g.

* `ROLL_PITCH_YAW_THROTTLE`
* `ROLL_PITCH_YAW_ALTITUDE`

Also, an alternative is that we change the names so they're shorter but less descriptive, e.g.

* `PASS_THROUGH`
* `RATES`
* `ATTITUDE`
* `ALTITUDE`

where's it described in the documentation that `ATTITUDE` is roll, pitch, yaw rate, throttle, etc. Do you have a preference?

superjax, 2016-05-30:
I'm thinking `active` was whether or not the channel was being controlled in that message (whether the other values should be ignored or not)

I think it's better to be more explicit in the code, so let's stay with the long names.  The compiler doesn't care how long the names are, and we have tab-complete.

dpkoch, 2016-05-31:
Works for me. That could potentially be folded into the `control_type_t` enum as well by adding a `NONE` value or something similar, then you could git rid of the bool. That might be slightly cleaner because if active is false then none of the `control_type_t` values are valid anyway, so you may as well just make it a different value in the enum. But either way would be okay.

Sounds good. I've opened a pull request with an updated `OFFBOARD_CONTROL` message, let me know if you like what I did with it or not. Don't merge it yet though, I'll add the callback and then you should be good to go.

dpkoch, 2016-05-31:
Okay I've added a simple callback. It just dumps the message into the global variable `_offboard_control`, which is a `mavlink_offboard_control_t` struct. It also saves the time at which the message was received in `_offboard_control_time` in case we want to check for timeout of commands.

superjax, 2016-06-01:
Alright, I've finished the first draft of the mux.  However, I can't use my normal printf stuff to test it.  Is there a chance you could help me design some MAVlink messages to test that the muxing is working properly?  I think all that I need is the 8 motor outputs sent over the MAVlink line and echoed on the computer side, I can hook up a transmitter and test the rest of it out.  That message might be useful for future testing anyway.

dpkoch, 2016-06-01:
So MAVlink already has a SERVO_OUTPUT_RAW message type, which lets you send 8 motor outputs (PWM 1000 to 2000 us values). The definition is at https://pixhawk.ethz.ch/mavlink/#SERVO_OUTPUT_RAW. Will that work? If so I'll add an optional MAVlink stream and a callback in fcu_io for it.

Also, just FYI, for debugging purposes there is a NAMED_VALUE_INT (https://pixhawk.ethz.ch/mavlink/#NAMED_VALUE_INT) message and a few other debugging type messages built into mavlink. There's also a STATUSTEXT message (https://pixhawk.ethz.ch/mavlink/#STATUSTEXT) that may be useful for doing things like reporting errors or acknowledging commands (it's big so should be used sparing and only when we're not flying). I imagine it would be useful to have the option to use those; if you agree I'll implement them as well.

superjax, 2016-06-01:
Those would be awesome!  For a first hack, I think just printing up the screen would be sufficient, I can see myself leveraging all of those messages you mentioned, but the servo output is probably the must critical.

Thanks man!

superjax, 2016-06-06:
Thanks Dan,  I'm going to close this issue!

closed 2016-06-06 by superjax