rosflight/rosflight_firmware #17: Offboard Control

superjax, 2016-06-13:
I think we are ready to start trying control from MAVlink.  We just need the hooks in fcu_io2 to send the MAVlink control messages.  @dpkoch, do you think you could help me get that set up?  I think I'm almost done getting the muxing between MAVlink and RC done, but i've been manually creating the control struct to make it work.

By the way, I wrote a latex document about the signal muxing.  

[rc.pdf](https://github.com/BYU-MAGICC/ROSflight2/files/312326/rc.pdf)

dpkoch, 2016-06-14:
Sounds good. Here's my proposed ROS message definition:

    # Offboard control command message
    
    # control mode flags
    uint8 MODE_PASS_THROUGH = 0
    uint8 MODE_ROLLRATE_PITCHRATE_YAWRATE_THROTTLE = 1
    uint8 MODE_ROLL_PITCH_YAWRATE_THROTTLE = 2
    uint8 MODE_ROLL_PITCH_YAWRATE_ALTITUDE = 3
    
    # ignore field bitmasks
    uint8 IGNORE_NONE = 0
    uint8 IGNORE_VALUE1 = 1
    uint8 IGNORE_VALUE2 = 2
    uint8 IGNORE_VALUE3 = 4
    uint8 IGNORE_VALUE4 = 8
    
    uint8 mode # offboard control mode for interpreting value fields
    uint8 ignore # bitmask for ignore specific setpoint values
    float32 value1
    float32 value2
    float32 value3
    float32 value4

I'm thinking that on the ROS side the values will be just in native units (rad, rad/s, m, or 0 to 1 for throttle), and then we'll do the conversion to int16 in the fcu_io node.

What are the units we're using to send the value over the wire as int16s? mrad, mrad/s, mm, 0 to 1000 for throttle?

dpkoch, 2016-06-14:
Alternatively we could have a different message/topic for each mode in order to hide the mode enum stuff from the user. I'm thinking that wouldn't be worth the mess though, and we still couldn't hide the ignore bitmask stuff anyway. So probably just keep it all in one message like this?

superjax, 2016-06-14:
I like this.  Let's go ahead and use it like this, and we can use floats, so let's do it in rad, rad/s and 0-1.  Then we'll do the conversion in the fcu_io node.

Awesome!  This will be cool.

dpkoch, 2016-06-14:
Alright I've opened a merge request in fcu_io2; let me know what you think. Also just a note for future reference, it might be better to open issues like this in the repo to which the changes will actually happen so we can reference issues in commit messages and things like that. Not a big deal though.

superjax, 2016-06-14:
Oh, right.  That makes sense.  Haha.

superjax, 2016-06-16:
I think this is awesome, was closed by #20

closed 2016-06-16 by superjax