rosflight/rosflight_firmware #18: MAVLink debug messages

dpkoch, 2016-06-14:
It'd be nice to have some generic messages available we could use for sending debug values over MAVLink and seeing them as ROS topics. Luckily, MAVLink has some messages already built in! The candidates are:
 1. [MEMORY_VECT](https://pixhawk.ethz.ch/mavlink/#MEMORY_VECT)
 2. [DEBUG_VECT](https://pixhawk.ethz.ch/mavlink/#DEBUG_VECT)
 3. [NAMED_VALUE_FLOAT](https://pixhawk.ethz.ch/mavlink/#NAMED_VALUE_FLOAT)
 4. [NAMED_VALUE_INT](https://pixhawk.ethz.ch/mavlink/#NAMED_VALUE_INT)
 5. [DEBUG](https://pixhawk.ethz.ch/mavlink/#DEBUG)

I don't think we'd use the vect messages much, so I'm thinking we won't add support for those. The named value messages are 9 bytes bigger than the debug message because they use a char[10] to identify what the value is as opposed to a uint8. However, `DEBUG` only supports a float, and the named value messages are still only 18 bytes and I don't think we're planning on sending them super fast.

So I propose adding support for the `NAMED_VALUE_INT` message type. If you think we'd use it, it'd be pretty easy to add the `NAMED_VALUE_FLOAT` message as well. What do you think @superjax?

superjax, 2016-06-14:
I agree with all of the above.

I did have an idea, though.  Is there just a generic string type?  I'm wondering if I can write a copy of printf that uses MAVlink instead of stdout, that way I can put "printf" statements, like I'm used to, but still use MAVLink under the hood.  All fcu_io would do is print the string to the screen.  It's slow, I know, but it's super super useful to be able to just drop in a print statement wherever you like.

What do you think?

dpkoch, 2016-06-14:
Yup, see issue #19. I meant to add that one when I was doing this one, just got distracted.

superjax, 2016-06-17:
I think I'm going to close this, since #19 is open.  The debug stuff has been working great for me so far.

closed 2016-06-17 by superjax