rosflight/rosflight_firmware #22: MAVLink timing

dpkoch, 2016-06-16:
Questions to answer:
- [ ] Is SerialWrite blocking or asynchronous?
- [ ] Is it buffered?
- [ ] What is the limiting factor on message rates? Do they start colliding or overwriting messages halfway through write?
- [ ] How to make the serial sending more robust based on the answers to the questions above. May include buffering or checks on whether the serial line is busy, etc.

nonrevlb, 2016-07-19:
I've been investigating the timing of the imu messages.

The timing of the naze is correct.  The following oscilloscope image shows it properly running at 500Hz.
[image]

However, Ros is receiving the messages in groups that are 4 or 6 milliseconds apart.  This is illustrated in the following histogram of the difference in ros_time for imu messages:
[image]

The source of the problem seems to be in the do_async_read function of mavlink_serial.cpp in fcu_io2.  I timed how often this function runs and got the following histogram:
[image]
Here we can see that do_async_run only runs every 4 or 6 ms.  Most of the function is very fast.  The line that takes up all the time is:
```
  serial_port_.async_read_some(
        boost::asio::buffer(read_buf_raw_, MAVLINK_SERIAL_READ_BUF_SIZE),
        boost::bind(
          &MavlinkSerial::async_read_end,
          this,
          boost::asio::placeholders::error,
          boost::asio::placeholders::bytes_transferred));
```

nonrevlb, 2016-07-19:
There is a reference to using a timer in conjunction with async_read_some [here](http://stackoverflow.com/questions/15385200/async-read-some-return) that may be helpful.

dpkoch, 2016-07-22:
See also https://github.com/BYU-MAGICC/fcu_io2/issues/25#issuecomment-234594569.

These are the results that I've been getting:
[image]
[image]

@nonrevlb while I've been unable to reproduce your results from above, I did also plot the timestamps reported from the naze, not just the time of arrival. What's interesting is that there are a few timestamps that report intervals bigger or smaller than 5ms. And they also seem to come in consecutive messages; one long and then one short. And they also seem to come pretty regularly once every second. To me this suggests that there might be something going on with the Naze firmware that causes the loop to slow down once a second. I'll start trying to dig into what that might be, but do you have any ideas @superjax?

superjax, 2016-07-23:
It's really hard to tell exactly what streams are happening in the naze, but the way we have it set up, it's likely that almost everything needs to be sent each second.

I did some timing testing on the controller branch, but I didn't look at max or min times for each function.  Maybe I'll get that going so we can know how long each function takes on average, and at maximum. 

Based on the way it is coded, I don't think that the main control loops will vary much in length from iteration to iteration.  There are some situations in which the estimator can run _faster_ than usual, but never slower.

We have learned that the mavlink stream function is really really fast, since is uses asynchronous reads and writes, however, the actual streaming may get bogged down from time to time.

[image]

Based on the way we have it set up right now, we will want to stream at these rates, which means we want to stream everything all at once 10 times per second.  I don't know if this could be causing an issue like this, but it is happening.

A similar issue may be happening with the i2cRead.  i2c is the most expensive thing that we do (just reading IMU alone takes almost twice as long as all the other functions combined).  Anyway, here is the "request map" for i2c read functions.  It also happens to line up with the MAVlink functions, so it may be that every once and while, we have a really difficult loop, because we need to update so many sensors all at once.
[image]

closed 2016-07-27 by superjax