rosflight/rosflight_firmware #28: Arming Logic

superjax, 2016-06-20:
We should re-code mode.c to be a state machine, rather than a series of if and else statements

superjax, 2016-08-17:
I looked into doing this, and I think that it would be more confusing that the if statements that we have.  There really are only four states.

Anyway, I think that it's fine as is.  I'm wondering if @dpkoch had insight that I don't have.

dpkoch, 2016-08-30:
#72 has a hotfix for issue where autopilot would arm on initialization if RC was not present. This issue should address this in a more sophisticated fashion.

superjax, 2017-03-17:
A lot of cleanup happened in #103 with the RC abstraction.  I think that we might still get more clarity from a state machine matrix approach, but until then, I think this is pretty good.

closed 2017-03-17 by superjax