rosflight/rosflight_firmware #35: New IMU data check is faulty

dpkoch, 2016-06-30:
@superjax Okay I could be wrong here, but this looks like it could be a problem.

On [line 37](https://github.com/BYU-MAGICC/ROSflight2/blob/master/src/sensors.c#L37) of sensors.c, we check the `mpuDataReady` flag to see if we have new IMU data available. However, as I was looking through drv_mpu6050.c, I realized that `mpuDataReady` never gets set to false, so after the first time that check will always return true. Is that correct, or did I miss something?

Two possible fixes:
- in sensors.c, set `mpuDataReady = false` like you do in your accelgyro example. Simple, but it seems a little weird to be modifying a flag inside a library from outside the library. Might be okay though.
- store a copy of mpuMeasurementTime and do an equality comparison. When a different value from your local copy is detected, you have a new measurement.

Thoughts? Also I wonder what this might have been doing to our timing.

superjax, 2016-06-30:
That's right. 

I just actually pulled a change to breezy where we register our own ISR - simondlevy/BreezySTM32#9.  That way, we have complete control of how the ISR works.  I got rid of the `mpuDataReady` flag entirely, and left it up to ROSflight to decide how to handle it.  It will be a lot cleaner, and it works better for camera triggering.

superjax, 2016-06-30:
testing

dpkoch, 2016-06-30:
Sweet! That'll be really nice.

dpkoch, 2016-07-07:
Closed by #36

closed 2016-07-07 by dpkoch