rosflight/rosflight_firmware #37: Time Stamp of IMU messages

dow611, 2016-07-01:
@superjax @dpkoch 
After zooming in on the IMU data I realized that they are not evenly spaced in time as they should be. The first plot shows the raw data, while the second is a histogram of the inter-message time. The mean is 100Hz but no messages arrive at that rate. This issue is evident on all of the flight tests we have done at the REEF these last couple weeks.

[image]

superjax, 2016-07-01:
That is interesting to notice how they seem to come in groups of two.    What is the x-axis of the histogram?

dow611, 2016-07-01:
Histogram of the inter-message time (seconds). We should see a spike at 0.01 only. The fact that there are 4 evenly spaced spikes is what's interesting to me.

superjax, 2016-07-01:
Yeah... that's kinda odd.  I'll look into this if I have time later today.  I can just turn stuff on incrementally and see what is making the stream rate go like that.  Are the timestamps from arrival times or header stamp times.  (Dan just implemented some stuff that should adjust header stamp times to be accurate)  I can confirm that the measurements are actually taken at the correct intervals.

superjax, 2016-07-01:
So, here is a sample which was run at 500 Hz using Dan's new timestamping
[image]

dow611, 2016-07-01:
I was typing this:

It appears like both the inter-arrival times using header time stamps and rosbag meta times are both exhibit this behavior.

Dan just left for lunch, but he said that the most recent firmware (as of yesterday) may fix this issue?

superjax, 2016-07-01:
Yeah, this is using today's firmware.

dow611, 2016-07-01:
Well, this basically means that we have been running our rmekf at 50Hz so maybe we'll get better performance now?

superjax, 2016-07-01:
Hopefully!  I don't know if Dan has told you, but we've set the default IMU stream rate to 500 Hz.  I still haven't seen any problems at 1000Hz, but to be safe, we dropped it to 500.

dow611, 2016-07-06:
We flew with IMU at 500Hz. This picture is zoomed in a typical imu acc y (similar for acc x) Units are m/s/s vs. sec
[image]

This is typical for gyros rad/s vs. sec
[image]

We don't know where that thick symmetric signal is coming from?! But it is saying we are bouncing back and forth between +/-5 m/s/s each dt!

The header inter-arrival times are pretty good
[image]

though ROS receives them in big clumps (metadata interarrival time is generally orders of magnitude faster)
[image]

The estimates actually looked pretty good
Attitude (RPY in degrees vs time in sec)
[image]

superjax, 2016-07-06:
Well I'm glad that the estimates look good!  It's super weird that you're getting that flipping signal.  I'm going to set up a naze on shredder so I can fly it and see what is going on...  I bet it's a problem with how we are handling the data, rather than the raw data itself.  Perhaps I'm wrong though...

superjax, 2016-07-06:
I assume you've balanced your props... haha

dow611, 2016-07-06:
They have never even heard of prop balancing here. :) Could that be it?

superjax, 2016-07-06:
Well, you'll only get better performance if you balance your props.  (the standard props that come with the 3DR quad kits are notorious for being out of balance - we generally use HQ props so we don't have to balance them),  Though that usually causes large amounts of noise in one axis, and tends to stay body-fixed.  I can't figure out why the noise would increase when you rolled.  I was making a joke more than anything.  I still think it's something that I've done wrong in the code.

dow611, 2016-07-06:
My first impression is that is is some type of aliasing. When we use it in a low pass filter (RMEKF) the noisy data still works, so the 'noise' is clearly correlated. This phenomena doesn't show up at all when we are sitting on the ground or when moving the IMU without flying.  
Could also be something to do with our gel pad we mounted the IMU to?

superjax, 2016-07-06:
Maybe... Do you have another IMU onboard that you can read, or are you flying the PixHawk?  I'm trying to see if I can see this behavior on our Microstrain, but it doesn't stream as fast, so It's a little hard to compare apples to apples.

superjax, 2016-07-06:
Here's a video on how to balance props [video](https://www.youtube.com/watch?v=4_fL_QuMkp8)

dow611, 2016-07-06:
Perhaps you fly shredder as is, but throw a Naze on as an external IMU and record?

superjax, 2016-07-06:
That's the plan.  I'm getting that ready as we speak

dpkoch, 2016-07-06:
For comparison, here's 4 seconds of data from the x-axis accelerometer on the Pixhawk:
[image]

The pattern does still seems to be there, but it's less well-defined and of significantly lower magnitude.

superjax, 2016-07-06:
[image]
Here is the plot of the naze vs microstrain.  I also put in a filtered version of the naze (4-part moving average).  I think that is likely why the data on the naze is so noisy.  It is completely unfiltered.

superjax, 2016-07-06:
Here is zoomed on on 4 seconds
[image]

Here is the filtered version for 4 seconds
[image]

superjax, 2016-07-06:
So, I'm working on getting the DMP running, to see what it looks like.  I don't really know what to do about this.  What do you think @dpkoch @dow611?

dow611, 2016-07-07:
dock_figure('Aliasing',10);
    t = 0:0.001:1000;
    y = sin(t);
    mask = 1:3100:length(t);
    tt = t(mask);
    yy = y(mask);
    plot(t,y); hold on;
    plot(tt,yy,'rx-')
    axis([0 1000 -2 2 ])`

[image]

Look familiar? Kevin suggested that it could be a high frequency oscillation (blue) that when we sample at 0.002 it creates our lobes. 

Since we already have a filter, we could 'filter' the data in the rmekf (like we just did) or we could do a moving average. The moving average would result in much smaller variance and a much more reaslistic value at any given time. The later makes sense to me. The measurements would become strongly correlated if we do a moving average and keep it at 500Hz so perhaps it would be best to publish the average of every 5 measurements (no overlap)? Thoughts?

superjax, 2016-07-07:
We can also set the LPF on the accelerometer itself.

[image]

Before I left work today, I set up the DMP to work within ROSflight.  After letting it simmer on the backburner this evening, I realized that perhaps I could pretty easily throw some soft rubber under the Naze and hopefully filter some of that high-frequency stuff out.  Looking at that graph above, it makes a lot of sense.

Tomorrow morning, I'll try running the DMP (limits us to 200Hz, but lots of data processing comes for free), then I'll try adding rubber, flying first it with the raw values, then again after changing the DLPF in the MPU.

dpkoch, 2016-07-07:
@dow611 That plot is very interesting; it looks very similar to what we're seeing. To me that makes prop vibrations seem like a pretty likely suspect; hopefully balancing the props will help. It's also possible that the green gel goo we have our naze mounted on is too stiff, or that the structure it's mounted on that also has a big computer strapped to it is contributing to the vibrations.

@superjax Yeah I think it'd be interesting to see what a light LPF does to the data. The DMP would be interesting to see, but long-term I think it's a no-go due to licensing issues. As far as filtering with the DLPF vs. a windowed average on the Naze, I'd say let's make the MPU6050 do the work if it gives us the performance we need?

superjax, 2016-07-07:
Here are some other ideas for vibration damping. [Link](http://ardupilot.org/copter/docs/common-vibration-damping.html)

superjax, 2016-07-07:
Okay, here are the results of my first investigation:

The DMP didn't do anything to help.  It looks like the DMP doesn't do any additional filtering beyond the built-in LPF register.

I did a little test with each of the LPF settings.  I set shredder on the cart, and armed the Pixhawk so the motors would spin. I didn't fly it, but I just wanted to see what the noise would be with the motors armed, because if it was too noisy just sitting there, then flying would definitely be bad.

[image]

So here are the results of each test.  Starting with the LPF off:
[image]
Then at 256 Hz
[image]
Then 188 Hz
[image]
Then 98 Hz
[image]
42 Hz
[image]
20 Hz
[image]
10 Hz
[image]
and 5 Hz
[image]

These are all plotted on the same scale (using the PyQtGraph engine in `rqt_plot` so it actually displays all the data - as opposed to the Matplotlib engine)

I think what I need to do next is fly some of the lower filter values.  I'm afraid that 5 Hz and 10Hz might be a little slow, but 20  or 42 might do the trick.  I did some digging, cleanflight uses 188Hz, while some power-users use 256 Hz, but those quads are generally way smaller than shredder or the 3DR quads, have smaller propellers (which cause less vibration), and faster dynamics.  I think that for shredder we'll probably want to drop down.  Especially with the results I just got.

dpkoch, 2016-07-07:
Cool deal. My only concern with filtering as aggressive as 20 or 42Hz is the delay (5 and 8 ms respectively). That seems like it's starting to get significant. Do you think 98Hz (3ms delay) would give us enough?

superjax, 2016-07-07:
Perhaps!  I haven't had time to test it yet, but after I eat real quick, I do a few flight tests and post the results.  I'll also try to record all the information we would need to run two estimators in parallel, one with the microstrain IMU and one with the Naze.  It might be interesting to see how the LPF affects performance once it goes through the RMEKF, because as David pointed out, this noise will not agree with the dynamics at all, so it will likely get filtered out.

dow611, 2016-07-07:
Thoughts:
- First, this seems like a great Dr McLain question. @superjax do you mind asking?
- If we know the delay and it's fairly constant (likely) then we can subtract it from the timestamp and the RMEKF will deal with it accordingly.
- The RMEKF is a low pass filter. Since we integrate the IMU in our dynamics, spikes high and low likely have limited negative consequences. Say we apply 10m/s/s on Ax for 0.002s and then -9.9m/s/s for 0.002s, etc. Overall, the vehicle is never allowed to 'take off' and in practice (given the one run we did as an example) it is as if the filter received the appropriate input (more or less). The crazy measurements are weighted by their dt in the filter.
- We might want to run a few more flight tests, but I leaning towards keeping it faster (freshest data is available) and forcing the RMEKF to deal with some of the noise. But I dunno?

superjax, 2016-07-07:
Okay, here are the results of my test.  I flew Shredder with the naze and the microstrain IMU at the same time, but varying the naze's LPF.

[image]
[image]
[image]
[image]

I don't really know what to make of it.  I have the bags though, so we could run it through RMEKF to see how performance changes
[naze_bags.zip](https://github.com/BYU-MAGICC/ROSflight2/files/353203/naze_bags.zip)

superjax, 2016-07-07:
Also, Dr. McLain is out of town.  He will be back on Monday, and I will definitely ask him.  It's very much up his alley.

superjax, 2016-07-12:
Spoke with Dr. McLain:

He thinks that we should look into better mechanical filtering (mounting it better) before performing any digital filtering.  He pointed out that the Kalman filter is a low-pass filter of a sort, but doesn't introduce delay like the LPF on the MPU6050, so it should probably perform pretty well even with high-frequency noise on the data.

He said to start, I should try to mount the naze right on the microstrain.  That way we are comparing apples to apples, but that it's probably not the sensor, because the naze isn't "making up" noise, it's just seeing more noise, and is capable of sending more of it, simply because it is read faster, and has worse mechanical filtering.

I sum, he didn't seem terribly concerned.  He sort of felt that if the estimates were good, then there wasn't much to worry about, and we could probably do a better job of mounting the naze and get better data.  He didn't think it was a good idea to do heavy LPF onboard the MPU6050, because the Kalman Filter would do a better job.

superjax, 2016-07-12:
Dr. McLain sent me this paper about this issue
[20160712_095217.pdf](https://github.com/BYU-MAGICC/ROSflight2/files/360213/20160712_095217.pdf)

closed 2016-07-27 by superjax