rosflight/rosflight_firmware #38: Move towards floating point *gasp* data representations?

dpkoch, 2016-07-05:
Hey @superjax 

So I'm just wondering if we want to take another step down the dark path you started us on by implementing your estimator with floating point arithmetic.

My question is this: should we just go for it and do all of our sensors (especially IMU) in floating point too? Here are some reasons why I think it may be a good ideal:
 - Doing the estimator in floating point didn't seem to be too much of a performance hit, so this might not either
 - Whatever performance we do lose, we could regain by going to the F3. Maybe loops run slower on the lesser boards, but if anybody wants serious performance they can upgrade
 - It would make the code __much__ more readable
 - I think it would simplify a lot of the issues we're seeing with getting this calibration stuff working. Right now we're trying to stream and calibrate in terms of raw values, but something seems to be getting lost in the translation. If we were just streaming everything over in SI units, figuring out some calibration parameters, then applying those on the Naze as SI units, we'd be much less likely to screw something up.

Obviously, the downside would be a speed hit.

At this point I'm just asking the question. What are your thoughts?

dpkoch, 2016-07-05:
Wait a second, you can get an F3 for $30? So why are we not just using them?...

superjax, 2016-07-06:
Hahaha.

Okay, so the reason we aren't using them is because I was stupid.

I thought that BreezySTM32 was only the F1 unit.  Turns out, it also has a library for an F3 unit.... 

I too, was scared of the floating-point stuff.  Like, I read somewhere that it would take 16 or 17 clock cycles to perform a floating-point multiplication and that scared me.  Turns out, 16 or 17 times 5-20 microseconds is still nothing, so who cares?

I'm on board, we can go ahead and switch the whole IMU thing to floating-point.

However....  Would we just then represent the data over MAVlink in float16?  I think that float32 would be a little bit overkill.

superjax, 2016-07-06:
I'll go ahead and try this, I'll do some timing stuff as well, and we'll see how it does.

superjax, 2016-07-06:
So, just a preliminary look around.  I think moving the accelerometer measurements to floating point will have a negligible effect on performance.  I don't think it'll solve #29.  I believe that is a problem that will require a pull request to Breezy to solve.  (It has something to do with the acc1G being returned from Breezy.....)  If we solve that problem, then I expect that all of these problems will go away.

However, we could just use float16 for the imu data, which would provide zero bandwitdth loss on MAVLink, and a super-negligible loss in precision per [this article](https://en.wikipedia.org/wiki/Half-precision_floating-point_format)

However, I don't think we want to change the mixer, because it works currently, and I don't want to mess with it.  It's pretty simple anyway, the rc, mux and mode all don't have multiplications in them at all, so it doesn't matter, but perhaps the controller should use floats.  (There is something in the controller, but it's completely untested and needs a lot of work).

superjax, 2016-07-06:
Just pulled out my F3 board, and Breezy doesn't work exactly.  I'm not sure if I should spend the time getting it all connected up right now.  I believe that we will need to make some adjustment to Breezy to make the F3 work.  The F3 library is in Breezy, but I think that the pins are laid out a little differently on the F3 boards - just enough that the accelerometer doesn't work, PWM is a little different, print statements aren't working, and I can't get the LEDs to flash.  I believe that it'll be pretty straight-forward to switch, but I think we can use the F1 boards for now, and then later, invest the time to get the F3 units working.

dpkoch, 2016-07-06:
Right, I think we need to figure out what's going on in the Breezy stuff. I guess I was just thinking that as long as acc1G is consistent with the actual FSR setting, then we could convert to SI units and it wouldn't matter as much whether it was 8g or 4g. But it would be best to just get the actual range that we want.

I think float16 could be pretty cool, but unfortunately it's not supported by MAVLink. But even if we change everything to float32s, we're still only looking at 28 bytes per message. Given that we've been able to stream our 16 byte message almost 4 times faster than what our normal use case is, my guess is that it's not going to be a problem. It still pains me a bit, but maybe it's worth just trying?

I think it's fine to leave the mixer as-is for now. If we decide to change it later fine, but let's just start with the sensor/estimation stuff and then change the controller over when we get around to working on it. Once we do the controller we can reevaluate the mixer, but I agree. Let's leave it for now.

Yeah sticking with the F1 for now should be fine I think. It looks like we're still not even maxing the F1 out, so if we are able to get everything working with floats on the F1, then cool. I agree, we can work on getting the F3 up later.

dpkoch, 2016-07-06:
Just FYI, I've got a start working on this. I haven't quite figured out supporting float params on the ROSflight side, but I think I'm close. I've been working on the temperature_float branch, as well as the accel_calibration_float branch in fcu_io2

dpkoch, 2016-07-07:
Okay I think I've got it working pretty well with floating point if you want to take a look @superjax. The IMU streaming maxes out at about 940Hz, so we do take a bit of a hit there sending everything as float32 values.

This still doesn't _solve_ #29, but it helps in some sense since everything now streams in SI units even though the scale is different onboard. The corrected scale isn't quite right on my rev2 board (it's about 0.3m/s^2 too big), but it's at least in the ballpark now.

What do you think?

dpkoch, 2016-07-07:
Here's the compensated accel readings as the IMU warms up (I don't have a compressed air can or a lighter):

[image]

dpkoch, 2016-07-07:
Timing results:
- `update_imu()` function: 585us, runs at 1000Hz (when not calibrating)
- `update_sensors()` function: max 588us, called at ~21000Hz

The 585us means we would max out at 1709Hz if nothing else was going on (or in other words if we want 1000Hz, 59% of our loop time is spent reading and correcting the IMU).

dpkoch, 2016-07-07:
As a side note, of those 585us spent in `update_imu()`, 562us are spent on the following three lines:
```
mpu6050_read_accel(accel_raw);
mpu6050_read_gyro(gyro_raw);
mpu6050_read_temperature(&imu_temp_raw);
```
which are the functions that do the actual reading of MPU6050 registers via I2C. This is where it would be really nice to have asynchronous reading done by the I2C peripheral with an interrupt to report completion, but I guess that's an issue for another day...

superjax, 2016-07-11:
Alright, alright, you've convinced me.  

I still kinda believe that it is a bit of a waste of bandwidth to send them as float32's, because we get no increase in accuracy.  However, I think it does make sense if we were going to use radios to connect the naze to the ground station, which could easily be the case, and it does seem a little weird to pass the `_accel_scale` across as a parameter.  I had some code working that did that, but it was a little wonky, because you had to call a `Imu::_set_scale` function to get the data where it needed to go.

Do you want to go ahead and merge `master` into the `temperature_float` branch, and I'll go ahead and merge it.

dpkoch, 2016-07-11:
Haha, okay cool. I agree, I'm not super happy about the wasted bandwidth; it'd be nice if mavlink support a float16 type. But it's a tradeoff, and I think the loss of bandwidth is worth the increase in clarity/reduced potential for error. Plus we can still get 500Hz no problem, which is stupid fast anyway. So yeah let's go this route for now, but in the future if we come up with a better way that helps with the bandwidth issue then I'm totally open to that.

Yep I'll get that merged in and open a pull request. There'll be a pull request in both ROSflight2 and fcu_io2.

superjax, 2016-07-11:
Cool I already merged the mavlink pull request too.

PR #48 (2016-07-11): Temperature float
Switch IMU data to floating point, both internally and over Mavlink. Closes #38, helps with #29.

Before merging, need to test the following:
- [x] parameter get/set interface for both integer and floating point values
- [x] accelerometer calibration, both in fcu_io and in ROSflight

I have to run but can do those tests in about an hour.
closed 2016-07-11 by superjax