rosflight/rosflight_firmware #39: Accelerometer z bias calibration incorrect

dpkoch, 2016-07-07:
Okay, I forgot about this, but this is another of the reasons I had for working on #38.

I don't know if you see this, but on my rev5 board (I'm not bothering with the rev2 board until #29 is resolved) the z accelerometer bias always gets calibrated incorrectly so that while sitting on my desk the z accelerometer reads ~8.75m/s^2. This happens with both the full temperature calibration done in fcu_io, and with the simple bias calibration done in ROSflight2. I do not see this problem with the floating point version.

@superjax Do you have this issue with yours? Any thoughts?

superjax, 2016-07-07:
Oh, I bet it's a sign error somewhere.  I'll look for it

superjax, 2016-07-07:
Found the bug.

```c
_accel_data[0] -= (_params.values[PARAM_ACC_X_TEMP_COMP]*_imu_temperature)/1000 + _params.values[PARAM_ACC_X_BIAS];
    _accel_data[1] -= (_params.values[PARAM_ACC_Y_TEMP_COMP]*_imu_temperature)/1000 + _params.values[PARAM_ACC_Y_BIAS];
    _accel_data[2] -= (_params.values[PARAM_ACC_Z_TEMP_COMP]*_imu_temperature)/1000 + _params.values[PARAM_ACC_X_BIAS];
    _gyro_data[0] -= _params.values[PARAM_GYRO_X_BIAS];
    _gyro_data[1] -= _params.values[PARAM_GYRO_Y_BIAS];
    _gyro_data[2] -= _params.values[PARAM_GYRO_Z_BIAS];
```

commit e37485a6 (2016-07-07): fixed #39
commit a1f0b7cd (2016-07-07): Merge pull request #41 from BYU-MAGICC/fix_temp_calibration

fixed #39
closed 2016-07-07 by dpkoch