rosflight/rosflight_firmware #47: Website

dpkoch, 2016-07-11:
Hey @superjax, so I was wondering if we wanted to pursue getting a BYU domains website for this project? I emailed and asked if they could set up a group domain for us, and this is the response I got:

> In order for us to set up a group BYU domains account, it will have to be sponsored by a full-time faculty or staff member and be "owned" by the department or college. If you find a sponsor for this website and it is "owned"  by the the department or college (or the MAGICC lab in this case) send us the following information and we can get this started for you.
>
> 1. Purpose of the shared account.
> 2. Full name of the sponsor of the account:
>   - (must be a full time employee/faculty member):
> 3. Preferred Net ID:
>   - (to be created for the group, not a person in the group,)
> 4. Preferred Domain Name:
> 5. Preferred Email Alias for the group:
>   - (should be an existing @byu.edu address for the group, if not, an alias will be created, this alias will point to the full-time sponsor)
>
> Notes:
>
> 1. The Preferred Net ID should not be an existing Net ID. A non-person Net ID will be created for the group, so the name of the Net ID to be created should be relevant and related to the name of the requesting group.
> 2. The Preferred Email Alias should be a previously created @byu.edu email address associated with the requesting group. If the group does not have an existing @byu.edu email address already, an @byu.edu alias should be created that is relevant and related to the requesting group.

What do you think, should we pursue this? Do you think Dr. McLain would be okay with sponsoring it, and do we want the website to be "owned" by the MAGICC lab?

superjax, 2016-07-11:
I'll add it to my list of things to talk with Dr. McLain about today.

I think that would be a great idea!  We would have the source code, and if there were ever a time that we wanted to privately host the website, there is no reason why we couldn't buy the domain from BYU, and host it ourselves.

Alternatively, we could use the byu-magicc.github.io (after we switch it to be all lower-case).  Which would be free, it wouldn't require Dr. McLain to forward emails and stuff, and would allow us to easily copy it somewhere else if we wanted to.  We would have to do a bit more legwork to make it look pretty, but I think it wouldn't be too hard to get a Wordpress or Bootstrap something going on the github.io site.

dpkoch, 2016-07-19:
@superjax Did you ever get a chance to ask Dr. McLain about this?

superjax, 2016-07-22:
I'm sorry, I forgot.  I'll have time tomorrow to talk to him about it.

So, are you more in favor of getting the rosflight.com domain, as opposed to using the groups github.io page?

dpkoch, 2016-09-01:
We got the rosflight.org domain through @dpkoch's personal BYU domains account.

closed 2016-09-01 by dpkoch