rosflight/rosflight_firmware #52: Gyro scale factor

dpkoch, 2016-07-18:
Something seems to be wonky with the gyro scale factor.

dpkoch, 2016-07-18:
Baseflight uses identical code for their scale factor (this is based on baseflight's driver):
https://github.com/multiwii/baseflight/blob/master/src/drv_mpu.c#L174

Cleanflight uses the following code (https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/accgyro_mpu6050.c#L80):

    gyro->scale = 1.0f / 16.4f;

which makes sense according to the datasheet if you want it units of degrees per second (see https://github.com/BYU-MAGICC/ROSflight2/issues/29#issuecomment-230694862). I still need to dig in and find out just how that value is used, but this makes a lot more sense to me (add in a factor of pi/180 convert to rad/s).

dpkoch, 2016-07-18:
Okay, so a little more on how cleanflight does it.

They use the `gyro->scale` value from my previous comment to compute a scale factor that puts it into rad/s (https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/imu.c#L147):

    gyroScale = gyro.scale * (M_PIf / 180.0f);  // gyro output scaled to rad per second

They then use that scale factor to pass into their estimator (https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/imu.c#L449):

    imuMahonyAHRSupdate(deltaT * 1e-6f,
                        gyroADC[X] * gyroScale, gyroADC[Y] * gyroScale, gyroADC[Z] * gyroScale,
                        useAcc, accSmooth[X], accSmooth[Y], accSmooth[Z],
                        useMag, magADC[X], magADC[Y], magADC[Z],
                        useYaw, rawYawError);

where `gyroADC` is the raw value read from the MPU6050 (or whichever IMU they're using) at https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/accgyro_mpu.c#L320:

    uint8_t data[6];

    bool ack = mpuConfiguration.read(mpuConfiguration.gyroReadXRegister, 6, data);
    if (!ack) {
        return false;
    }

    gyroADC[0] = (int16_t)((data[0] << 8) | data[1]);
    gyroADC[1] = (int16_t)((data[2] << 8) | data[3]);
    gyroADC[2] = (int16_t)((data[4] << 8) | data[5]);

So in other words, yes. Cleanflight uses the scale factor that we think makes sense from the datasheet.

So I think that we would be justified in changing Line 220 of `drv_mpu6050.c` to

    // 16.4 dps/lsb scalefactor for all Invensense devices
    *gyroScale = (1.0f / 16.4f) * (M_PI / 180.0f);

and opening a pull request to simondlevy/BreezySTM32, the result being that the gyro scale value returned by `mpu6050_init()` converts the raw data into units of rad/s.

Thoughts @superjax?

superjax, 2016-07-18:
I've hard-coded the scale factor to be `*gyroScale = (1.0f / 16.4f) * (M_PI / 180.0f);`, and I've been putting it through the estimator with it only integrating gyros, not using the accelerometer at all.  Oddly, I'm getting it estimating 1/4 the angles that I rotate it...

superjax, 2016-07-18:
I wish I had a better way to measure the actual angular velocity.  Doing it by hand is pretty terrible.

superjax, 2016-07-18:
wait... hold on, I think I might have tested it without saving the cpp file... 

I just tried it again, and it worked.   That was stupid of me.  Yeah. Let's pull a change to Breezy that fixes that.

superjax, 2016-07-19:
So, I think that the `4.0f` is important.  I get _horrible_ performance out of the estimator without the 4.0f.  The estimator does fantastic in SIL, so I know it's not the estimator by itself.  I think that the actual scale factor is 

`*gyroScale = (4.0f / 16.4f) * (M_PI / 180.0f);`

Why that is is a total mystery.  Based on the datasheet, the `4.0f` doesn't make sense.

dpkoch, 2016-07-19:
Well, I've discovered the source for the ever-elusive 4! Check it out: (https://github.com/simondlevy/BreezySTM32/blob/master/drv_mpu6050.c#L267)
```
void mpu6050_read_gyro(int16_t *gyroData)
{
    uint8_t buf[6];

    mpuReadRegisterI2C(MPU_RA_GYRO_XOUT_H, buf, 6);

    gyroData[0] = (int16_t)((buf[0] << 8) | buf[1]) / 4;
    gyroData[1] = (int16_t)((buf[2] << 8) | buf[3]) / 4;
    gyroData[2] = (int16_t)((buf[4] << 8) | buf[5]) / 4;
}
```

So it divides by 4 in the `mpu6050_read_gyro()` function, then multiplies by four with the gyro scale factor!

???????

Leftover goodness from the ever-wonderful baseflight code...

superjax, 2016-07-19:
that is straight-up retarded

dpkoch, 2016-07-19:
Haha yeah. Agreed.

I can go ahead and fork BreezySTM32 and open a pull request to Simon

superjax, 2016-07-19:
Cool.  Sounds good!

dpkoch, 2016-07-19:
Will be resolved by simondlevy/BreezySTM32#11

PR #58 (2016-07-20): Gyro scale
Closes #52
closed 2016-07-20 by superjax