rosflight/rosflight_firmware #59: Board support layer

dpkoch, 2016-07-20:
So I think this SIL stuff could be really cool; it'd be good both for people who just want to use the firmware as is but need a simulation option, and it would really open it up for people who want to do things like develop and implement new attitude controllers on the autopilot itself.

One challenge with the current implementation though, is that it requires duplicating a lot of code between ROSflight2 and the fcu_sim plugin in order to change the low-level functionality to work with gazebo instead of BreezySTM32.

A potential solution would be to implement a board support layer (BSL) that abstracts away the serial communication, sensors, actuators, and memory aspects of the firmware. The actual autopilot code that is not board-dependent would then be separated out and sit on top of that.

For our case, this means that we could have BreezySTM32 BSL implementation and a ROS/gazebo BSL implementation that would be interchangeable without having to modify any of the higher level functionality. Depending on how we structure things, we could even have the ROS/gazebo BSL implementation live inside of the gazebo plugin so we don't clutter up our firmware.

In fcu_io, we could also have two different interfaces: one that uses an actual serial port to pass mavlink messages back and forth (the current implementation), and one that uses ROS to pass mavlink messages back and forth. The ROS/gazebo BSL implementation would then use that interface instead of the SerialWrite from BreezySTM32. Then since both versions go through fcu_io, the ROS interface would be absolutely _identical_ between hardware and sim.

In summary, I think it could look something like this:
[image]

There are a few ways we could implement it. One example, using C++, is given once again by @simondlevy at https://github.com/simondlevy/hackflight. If we want to stick with plain C, another option would be to just create a board.h header file with function prototypes, then for each case create a board.c/cpp file that implements those functions and then include it in the sources list. There would definitely be some technical issues to iron out, but it could potentially be pretty slick.

Thoughts @superjax?

superjax, 2016-07-20:
I would be totally on board.  I wouldn't be opposed to going to c++, since that would make things a lot easier, but moving to C++ would probably require a huge re-vamp of our stuff.  Perhaps let's not do that quite yet.

What if instead we used the board.h header like you said above?  That seems the most straight-forward.  This also would make it easier to switch to the F3 or F4 chips when we are ready.  I'm on board, and it shouldn't be too bad.  Let's get the controller working first.  I think we are about there.  I'll go test it right now, and if it works, I'll do a pull request.

simondlevy, 2016-07-20:
It sounds like you guys are really forging ahead with this project!  I'm especially interested in seeing how you work with the new F3 and F4 flight-control boards.  I tried working F3 support into BreezySTM32 but gave up after realizing I was in over my head.

As for going to C++, I'm not sure whether you mean switching to C++ from straight C, or going to C++ from a completely different language.  If it's just C -> C++, I wouldn't worry too much: my C++ firmware in Hackflight is really just C with a thin wrapper of C++ classes.  Looking back on it, I could just have easily stuck with the original Baselfight approach: a bunch of C modules with static variables and methods for their "private" components, and extern for the "public".  Since I'm not creating more than one instance of anything, C++ provides no real advantage.  I had to wrap everything in "extern C {" to get it to work with Breezy STM32.

dpkoch, 2016-07-21:
Hi @simondlevy, thanks for the input! We'll keep you posted on the progress with the F3/F4 boards, but that's probably not going to happen until a little later down the line.

We were talking about going the C -> C++ route. It's good to get your insight having tried it out. I think the only reason we would do it is to make the code more readable and a little better organized. We may make the switch at some point, but the C version really isn't too bad so we will probably just push ahead with that for now.

superjax, 2017-03-18:
Completed by #103

closed 2017-03-18 by superjax