rosflight/rosflight_firmware #61: RC Trim Saving

superjax, 2016-07-26:
We need a way to save trims from the RC transmitter into the FCU so that ROS is controlling with respect to trimmed commands

dpkoch, 2016-07-27:
We could add a command to the MAV_CMD enum for saving current RC values as the trim values. So you'd fly, trim everything out, land, disarm, then leave the sticks centered and send a mavlink command message to save them off?

One tricky case would be when the transmitter and offboard control are in different modes. If you trim everything out with the transmitter in angle mode, then send offboard commands in rates mode, then what do you do? You still need to trim things out, but the RC trim values no longer apply to the offboard control inputs.

It doesn't seem like we've needed to do this with the Pixhawk. Any ideas why? Could we solve the problem just by adding integrator to the controllers (I'll try to get that put in today, sorry for taking so long)?

superjax, 2016-07-27:
The reason we need to be able to do this is to account for things like the center of mass not being in the center, and differences in performance from different motors/propellers.  It actually is a problem with the PixHawk, but we have solved it on shredder by adding weight to the prop guard to balance it out.  We could always do that, but honestly, that's a stupid idea, since we can't 'balance out' yaw instability, and it's better to just figure out the 'equilibrium forces' and add them to the control output.  Using integrators is a potential solution, but they won't be as clean as using equilibrium forces.  On newer versions of the PixHawk firmware, there is a method to save the RC trims into the autopilot so it does it automatically, without needing extra weight to balance.

We have also never flown the Pixhawk in rate mode from offboard control (I don't think anyone has), which is probably why we haven't ever run into that before.  But you're right, depending on the mode, you'll have different trims and you'll have to save them appropriately.

So far, I have added a MAV_CMD_DO_RC_CALIBRATION, which saves off the center and range for each channel.  It then calculates (in SI units) the additional control that center sticks represents, and appends that control to the `_offboard_control ` struct.

My only decision left is to decide how to deal with the `_rc_control` struct.  We have two options: the first is to do it the same way that the PixHawk does it, which is after you have completed an RC calibration, the autopilot saves the trims, and you are supposed to manually zero out the trims on your transmitter.  My hesitation is that that procedure is a little confusing.  The second would be to not add the equilibrium force to the `_rc_control` struct, and instead leave it alone, and let the trims in the RC transmitter take care of it.  It is less clean in the autopilot code, since the equilibrium force then lives on the RC transmitter, but I think that for a user, it is less confusing.

What do you think?

superjax, 2016-07-27:
Closed by #62

closed 2016-08-02 by superjax