rosflight/rosflight_firmware #74: Altitude Complementary Filter

superjax, 2016-08-18:
We need to derive a complementary filter for altitude in order to perform altitude control.  

I haven't found any good references for this yet, unfortunately.  If I find some, I'll put it up.

superjax, 2016-08-18:
Still haven't found a good reference, however, thinking about this during the IAB meeting, I have come to the following conclusion.

We should propagate our altitude using accelerometers.  We have a quaternion of our attitude, so it's pretty straight-forward to rotate our acceleration into the inertial frame, and integrate the inertial acceleration vector.  This will be incredibly noisy and drifty, so we need to filter it.

We can use sonar and barometer to both filter the integrated acceleration and estimate accelerometer biases.  We can only estimate bias on the inertial z-axis, but if we are frequently changing our pitch and roll, then we should eventually converge on the accelerometer biases for all axes.

What do you think?  I'll talk to Dr. Beard about it, and think about it on the plane.  I think it might work.  We should at least be able to get a good estimate of our z-bias.  the x and y biases might not converge exactly, but it's an interesting idea.

dpkoch, 2016-08-21:
Yeah, that's more or less what I've been thinking in the back of my mind. I would think it should work pretty well. Barometer is a bad idea for multirotors; the prop wash wreaks havoc with the measurements. That's what we were trying when the backflip video happened. It would be good for fixed wing, we just need to make sure it's turned off for multirotor.

dpkoch, 2016-08-21:
We could also estimate only the z accelerometer biase and leave x and y be (assume they're zero). I would imagine they're going to be pretty close to unobservable under normal operating conditions, but they should be pretty small from the calibration and they should also have not much effect on the estimates since they'll always be multiplied by the sine of a small angle.

superjax, 2016-08-21:
Do you know the formula to rotate a vector by a quaternion?  If I remember, it's the sandwich product, right? 

` v' = q*v*q'`.  Where `v` and `v'` are the pure quaternion created from the vector. `(0, x, y, z)`

superjax, 2016-08-21:
I just found a simplified version that avoids the two quaternion multiplies!

https://blog.molecular-matters.com/2013/05/24/a-faster-quaternion-vector-multiplication/

```c
t = 2 * cross(q.xyz, v)
v' = v + q.w * t + cross(q.xyz, t)
```

closed 2017-03-17 by superjax