rosflight/rosflight_firmware #78: Releases

dpkoch, 2016-09-14:
@superjax I was thinking we might be at the point where we want to start creating releases. That would allow us to define points in the code where we think there's a stable and substantial change for people to upgrade to. In addition, we could attach the compiled firmware binary to the release so people don't have to build it themselves, just flash it.

I propose releasing our current code in the master branch as v0.1.0, since it's kind of the first pre-release where most of the functionality works. What do you think? If you approve I'll give it a try.

superjax, 2016-09-15:
That's a really good idea!  Go for it.  It's pretty easy to manage releases, and, as you said, it makes it easier for others who may not be savvy enough to build it themselves.

superjax, 2016-09-15:
Should we perhaps nail down a features list for v 0.2?  I'll start a new Milestone.

dpkoch, 2016-09-19:
Yeah sounds good, although I'm not sure we'll have features so much as cleanup tasks.

Also, the reason I haven't done this yet is because of #80; I don't think we should create a release until that's resolved. I'm pretty stumped on it at the moment though.

superjax, 2016-09-20:
Agreed.  Okay, let's get #80 fixed, then we will release.

superjax, 2017-06-27:
This was completed by #90

closed 2017-06-27 by superjax