rosflight/rosflight_firmware #80: Param write broken

dpkoch, 2016-09-17:
Parameter writing doesn't seem to complete; the light doesn't flash, no command acknowledgement is sent over MAVLink, and parameters aren't saved

dpkoch, 2016-09-17:
I've been trying to track down the problem, but I'm kind of stumped. It appears that the breakdown is in the `bool writeEEPROM(bool blink)` function, in this `for` loop:

    for (unsigned int tries = 3; tries; tries--)
    {
      mavlink_log_warning("Tries remaining: %d", tries);
      FLASH_ClearFlag(FLASH_FLAG_EOP | FLASH_FLAG_PGERR | FLASH_FLAG_WRPRTERR);

      FLASH_ErasePage(FLASH_WRITE_ADDR);
      status = FLASH_ErasePage(FLASH_WRITE_ADDR + FLASH_PAGE_SIZE);
      for (unsigned int i = 0; i < sizeof(params_t) && status == FLASH_COMPLETE; i += 4)
        status = FLASH_ProgramWord(FLASH_WRITE_ADDR + i, *(uint32_t *)((char *)&_params + i));
      if (status == FLASH_COMPLETE)
        break;
    }
    FLASH_Lock();

If the write works, it runs straight through as expected. When it doesn't work though, it goes through all 3 attempts but on the 3rd attempt seems to never exit from the inner `for` loop. It doesn't freeze, because the autopilot keeps doing everything else, but it's as if it skips executing the rest of the function without actually returning. Very weird. Any thoughts @superjax or @gellings?

superjax, 2016-09-20:
That is really weird.  I'll see if I can't work on it this weekend.  I've been trying to get the asynchronous stuff working, and been running into issues myself.  Perhaps it'll be good to work on something else for a minute.

superjax, 2016-10-12:
Was this solved?

dpkoch, 2016-10-12:
Nope. I've been trying to narrow down the commit where the bug was introduced, but haven't been able to yet. There doesn't seem to be a single commit where after that commit the issue always exists. It seems like instead it might happen in one commit, but not the next, but then comes back again a couple of commits later. So I haven't had any luck tracking it down yet, unfortunately

dpkoch, 2016-11-07:
One thing to try might be disabling all interrupts before we try to access the memory? Don't know if this would solve it, but it's possible that entering an interrupt routine messes with the EEPROM access somehow. It may also explain why it used to work then stopped, since we didn't use interrupts before then started using the IMU interrupt. Maybe? Just an ideal I had; I'll try it out as soon as I get a chance.

superjax, 2017-01-20:
This is fixed as of commit 2c879f8

closed 2017-01-20 by superjax