rosflight/rosflight_firmware #85: F4 support

superjax, 2016-11-15:
I just wanted to write an update for my progress in building F4 support.  I've bought a revo from ready-to-fly-quads [link](http://www.readytoflyquads.com/flip32-f4).  And I started by trying to strip the relevant drivers from cleanflight, betaflight and raceflight

This was a horrible experience.  I can't believe that cleanflight labels itself as ''clean''.  There are some absolutely terrible coding practices going on in cleanflight, and after about 20 hours, I gave up, and decided to write my own.  In about an hour, I started getting some functional builds....  I feel stupid for even trying to do anything with cleanflight.

The biggest difference, besides the faster processor, is native USB support.  I had never worked with a USB peripheral on an embedded controller before, so that was something new.  I was hoping that I could bypass learning about how it works by relying on cleanflight, but as I already mentioned, turned out to be a terrible idea.  It's not all that complicated, but I need to make sure that it is using asynchronous callback methods or DMA or something like that.  I believe that we will be able to have even higher bandwidth on the VCP, (virtual COM port) than we have seen previously.

Anyway, now that I have decided to completely leave the stupid flight XXXXflights behind, I think I'll have a working solution in a couple of hours.  I intend on replicating all the BreezySTM32 examples with an F4, but I'll probably leave it as a separate repository just so it doesn't get complicated

superjax, 2017-02-27:
Hey @drinkdhmo, thanks for volunteering to help with this.

Here is my Revo.ioc file from STM32CubeMX.  I am pretty sure that I probably messed it up, so it might be worthwhile to just start over, but if it were helpful I just wanted to make sure you had it.

[link](https://magiccvs.byu.edu/~james_jackson/Revo.ioc)

superjax, 2018-01-18:
Okay, I have this working.  At the very least, communication seems to be working nicely.  Things still to do

- [ ] Confirm communication is working
- [ ] Confirm all sensors are functioning properly
- [ ] Long bench test to ensure no errors crop up
- [ ] Confirm RC and motor outputs are functioning properly
- [ ] Fly on multirotor
- [ ] Fly on fixed-wing

atyshka, 2018-05-24:
Is it planned to move the F4 documentation onto rosflight.org? Just a suggestion because checking out this library I had no idea it supported anything other than F1 controllers until going through the issues.

dpkoch, 2018-05-24:
Hi @atyshka, thanks for the suggestion. F4 support is still under development and hasn't yet been released (should happen soon! Relevant pull request is #278), but once that happens then yep we plan to move the documentation over.

PR #278 (2018-06-20): F4 support
Closes #85

Still Needs:

- [x] Airspeed Driver Support
- [x] Documentation Update
- [x] EEPROM check
- [ ] Fixed Wing Flight Test
- [x] Fly on a Multirotor
- [x] I2C Sonar support
closed 2018-06-20 by dpkoch