rosflight/rosflight_firmware #89: timeout error greater than 32 bit range

danems11, 2017-01-19:


superjax, 2017-01-20:
This is a pretty serious bug.  The micros timer rolls over after 70 minutes.  The solution is to convert the micros() function to return a 64-bit integer, which would roll over in 73,000 years!

superjax, 2017-01-20:
@danems11, I just pushed this fix to the latest polling branch.  Would you give it a try?  It turns out that the micros timer resets after 70 minutes.  Since you guys are using this for a long period of time, would you mind watching the time on the IMU message to make sure that it doesn't reset after about 70 minutes?

danems11, 2017-01-23:
We will test it out. Thank you.

superjax, 2017-01-26:
I'm going to close this for now.  I suspect that it is working with that previous commit

commit 5428b52c (2017-01-20): elevated micros timer to 64-bit.  Should fix #89
closed 2017-01-26 by superjax