rosflight/rosflight_firmware #102: Exciting Earth-Plummet Destruction

superjax, 2017-02-21:
We had another occurance today of no control from RC.  It looks like the naze rebooted twice in flight and was receiving failsafe commands (full throttle, level wings and elevator) from the RC receiver.  It looks very 
much like brown outs on the receiver.

I suspect that we pulled too much current on our 5V rail  here's the math

9g servos can draw 700mA [link](http://www.wattflyer.com/forums/showthread.php?t=59485)
the Pi has been seen to pull 700mA [link](https://www.pidramble.com/wiki/benchmarks/power-consumption)
I'm going to assume both the Rx and Naze each pull 500mA.

```
  4 x 700
+ 1 x 700
+ 2 x 500
---------
     4500
```
We are using the cheapest 5V 5A BEC of all time, so if it actually dies around 3A, then we would totally have brownouts.

el3ment, 2017-02-21:
data:image/jpeg;base64,/9j/4AAQSkZJRgABAQAAAQABAAD/2wCEAAkGBxMTEhUSExMWFRUWFRcXFxYXFxUXFxcVFxUWFhUVFxcYHSggGBolHRUVITEhJSkrLi4uFx8zODMtNygtLisBCgoKDg0OGBAQGi0fHx0tLS0tLS0tLS0tLS0tLS0tLS0tLS0tLS0tLS0tLS0tLSstLS0tLS0tLSstLS0tLS0tLf/AABEIAMIBAwMBIgACEQEDEQH/xAAcAAABBQEBAQAAAAAAAAAAAAADAAECBAUGBwj/xAA9EAABAwEFBQUGBQMEAwEAAAABAAIRAwQSITFBBQZRYZETcYGhsSIyUsHR8BRCcuHxB2KSI0OCohYzshX/xAAYAQADAQEAAAAAAAAAAAAAAAAAAQIDBP/EACQRAAICAgIBBQEBAQAAAAAAAAABAhEDEiExBBMiMkFRYUIj/9oADAMBAAIRAxEAPwD0CgzJaTGYZKrTZjC0KOIWqMhNauF3xs120E/E0O8fdPou9XMb9UJFN/Mt64j59VM17SkcUWqvaGySRhJlXCEFzFzM0OX3poTTDvhcPPBck8L0DblnvUXjlPiMfkuAchFAKoxHcul3YfLCODvULnKmXitjdWr7bm8RPT+UPoDetLMThzj77lzNem+kyrECHhzcATB9mQDiNMV2FsZiDxC5e2UYrvESHUjhlJaJGPexERMsW5ofSJxlzA7oQ70JRdl1SHUiREtbOGJxLc89Am2W/tKTHCGxLczhMtwzORCr7NqywF95zmvcPe4gOE6/lKYHU7x0L9mqDg28P+Jn5LzekvWGtD6fJzY8CF5SWXXlvAx0MJLopdkyERwyPEKN1Xqw/wBCmdQXNOWhvCDnqkzRFNvBSBKQCQiUASXeboVb1maPhLm9DI8iuCC67cWrhVp8CHdRB9Aqa4M5HU3U0IkKKRBAhRIUyVBzkxEqNS64O4EEeBle103ggEZET1XhxK9e3WtF+yUXcGBuRzb7J78s1aA1oUXBSCYqwBuCG5qK5DcmKgBYnRLpSRYqMbKO9XKQwVQiW4cirFD6FNMQcBZW89lDrO+M2kO6Z+S1lG00r7HNOrSOoVNWgR5S8IRardWnBhBLVyM0RTq05BB1wXmVtpXXObwJC9UqNXAb12e7Xd/cAURKMI4g9Vc2FUu1mczHVVgFGzvgg8D6FUB6HaG+wDw/hYNppVHvZcBIDvaAwBbhM8cyuks8OYNQQD1RmNhRFg0c9snd+o1ha5wAJJEZgYeeC17HsGkyTE3jeM8cdPErQpo7QqsKGZSAAAwAXmO8NmuWmo3+6f8AIXvmvUoXB790Irtf8TB1BI+iI9gzBaMMU7XGI8U9NqmxuKGiwQCk1qIGeqIGJ0BBrVt7o1btoI+NhHiMfqsq6tfd3Z1Z9am6nSe8B2Ja0kAZGSBAzTolnayoytyy7q2l+bQz9R+QWtZtym/7lUnk0AeZlGpNHGEpU6DnGGtc48ACfRek2XduzM/2w48X+164eS06VJrRDQAOAEDyT1DU83s269pf+S6OLzHln5Lvd26LrPRZReQ6CcRP5nExj3lXlS2hazTLHQILwD46p9BRuSkqrHlFammKiaYhIBOVQiF1JShJAGJZj7MeCJQVay1QcRk4Aj76K1SEdUIksqTVGVNquwPOtuWe5XqN/uJ6+181mkLpd8aEVg74mjqMPSFzxC5ZdloruauO34s8Fj+RB9Qu1LVg742a9ZyfhId8io+ykedOzQzmjFQFMuMNBJ4ASVqkDO83crXqLOQu9PsLUXObq1C1rmEEEOBg8/4W+SorkGWGEIzXKqxadi2XWqe5Se4cYMdTgroQCVzO+9C9TY/4XEdR+y9Hsm59d3vFjPEuPQfVaR/p/Z3tu1nPeMDANwYd2PmhLkdNnz3QatXZu79prEdlQqP5hhj/ACOHmvoLZu69hs8dlZ6QI1u33f5OkrZDuDT44KqRaR4hs7+lNtqY1DTog/E684f8W4ea6rZv9I6DYNas+pxDQKbT6nzXoxB5Dz++ibs+JJ8fonwBz9h3NsFAgss7J4vl57/bJW5TAAhow4AQPooivTGF4TqBiegxSFoJ92m48zDR54+SACieAT3T94IBNU/A0dxcfkFH8KT71R57iGj/AKAIAM8tbiSB3n6oB2gzQl36WucOowUmWKmDIY2eJEnqcUeECKjrTUPu0vF7gPISsnb76oYHOLLt4ey1p63ieXBb9VwaJcQBxJAWJvFaqZouF4EmIjHEGc8lD6JZ0NF95odxAPkjXgszZNT/AEaf6G+iuX0tqHQY1EjUVe/zQa1ra3MhPcVFw1EllHaTPiHVJT6gUYm61W9ZLM860mjv9kD5LdCyd3LGadlo03ZtYJ5HOPNahwjvWt8maLbEZqr0XLM3r25+Fo3h77jdZPHU84+YVqSoKZV33uhjHFwBBIgkTB4DXJcaLSwmA4LEtm0i9xe9znOOZOJVGrbgNFhOePts0WLI+kdZCz9qta5jqZ/M0jrqsahtqNSEZ9ra/GceR+qmCjJ8OwlGce0ZNn2FRZi6XnnF3pr4qW2K12i8MF3DICMNcldqNnI64k6DitCz7pvtALWV6DpwjtCHY/2ls+S2ba4FHuzlNyqZqWunQvQKrrkxMYEgx4ea9rsm5NnZjUc5/ebo6DHzXB7F3Wbs20NrV3tq1GTcp0yfZJEB1RxGGBwAC6C1b41fyBjO4SepUqJtknBtHZUrJZaAkMpt5kAf9nfVQfvLZwQO0b/2P/yPmvLbVbnPcXOcXE4knimpVJ5YnHorSRluz2ig++0Oa8FpEgtiCEqhY333AfqP1XNblntKJBLoESA4gEkuBMDjdBT72bwUtnNpO7G8ary2RDYiCSTGOak0vizovxrPyhzv0tMf5HDzSFoqHKmB+p3ybPqvObRvZan/AJw39LR6mSsTeHeis2y1aJqVS6rcLXhzvZuPaXgkGQC2VWrI3R6/2dQ51I/S0DzdKZtkbPtS7D8xLhhyOHksvYG2Q+x0azyJNMXnOc1oLm+yTJ4kLTslrbVF5j2kTEtk48JUllxrAMhCToGaCZ4n77lEMHBABDXb390n0UDW4NPjCRCgSEWFDVHvORA8JPmfkhPozm5/g4t/+YTvrjigvtPBKxUO2ysGN0TxOJ6lUN46INndGhacO+PmrnaOOWShUplwLXYg6fwkyHRnbCtNUUWgiIkAnOJMYK861P8Ai6D6qQpjLD9lMxqp0Cyo6sXZhx7z9E7GjKAD3T6qzI0HkoPMjUI1J2IXT8TvJJN2x4HwST1QbGhTGCg/LoptGaA3Mg8fJOxJFqm9cv8A1C2fVr06RpNv3S6QM8QII45FbFW1NYbpa52GQEjqYCapbXugNpho/udj0aPmpl7lTLjLV2eWjdu1nKzv8YHqVzW0Q9jiyo1zHDRzSCvdAKhzfHJoA9ZKr23YlGuIrM7Tm4kkdxnBZrDE39eTPB+0+wnbW4Fd7t/+mpbL7PUBHwPIDu5rsj4x4rgLfs+rScW1GOaeYIKzliRpHMyzSt7hqrlHavLxC5++4aT6qTbQO481NZYdO0V/yn2qZ2FPajXZme/Aoj7p90/PzXItqo9K1OGRVR8priSFLxE/izqrPY3PgBzJPF13zOC2LPuzaAJd2YB+KowfNcXQ2q4Z4rRs22G6iFrHyMbfdGMvGnH6s9a3Wpsos7IVqTnnEhhBy0GOOJOPNYX9XbIK1kaGvDqtOq1zWXmhxBBa6B4z4LlaFuYSC1xacwQcjxGoR6gvEuLi4kkmImTiTn8l0xSb4ZjJtKmijRv3WgkA3ROuMY8kQ4gtMOBBBkDEHMKd9nA9U/bN+EeZ+a0oyItdAA0GQ0HcNF1m520OzBBynEcozHPBcr+I5Don/FH7hJxGnTs9b/HCMMVA2lxyELykWx3FWLPtao04PKj0/wCmnqnpZe45nzUSxchYN6XZPAPr1W/Z9pse28CI5nLvCTjQ1NMvwEsP2Cy622KQzrMHcWlUq28tnH+453cHQkJs6B1QD7hMavcuTfvXRb7rHuPGAPMlVKm+FQn2abW/qcT6AJktnaGpH8KBrrgrRvFaXfmaz9LR85VKvtCs73qrj/yj0SonY9Eq2qMyAOJgeqo2jbVFvvVm9wM+i4C/jx804fhqnQrO0/8AI7P8Z/xckuIDzx++iZFAez0zqqdu2jQpn/UqMb3uEyqv/wCmAF5RvSALVULcQ514Rj72Jy5ysYu+C3wehbQ3wsjYh5eRPutJ1wxMBY1o/qA0f+uge9zgPIArhG03nJpV+y7DtFT3WHxWmqJs2q+/FpdlcYP7WyeriVc3f3mqOqXK1Rzg7IzEHw0VOxblVD77w3kMSt6x7p2WnBqGY1e+6OghDopWdHSYImFnbx17MGXK7WvkYMzdjqDm3vVLbW3mUv8AToy510e0QQB4HErirdazi5xJJ45kqaVbS6Ltt1Hso7V2Y1xmmIAya4yf8vqsC12SMHCDzz8CtSraXHVVqjpzxXJLy437UdkPFlXuZiPpuafZJ7k7LUR7wIVytTnJViw6+f7K1OE/kJwyY/iHp1gcijNqLKqU+AIPEfspU6tQYReHRRPAv8s0hnfUka7ayt0be8ZH5rLpSeSs0oGoXNTj0dD1fyNP8c86+QUTaXfEVU7Zo1HVNTtAdMGY106pOWV8tsSjiXCo2LPb9Hdfqj/jWfEsSUxJW0PMyJV2Yz8SDd9G7+NZ8SX4tnxBYMlMCVa86f4iH4Uf1nRttA0IVyz2yOY6yOB4hcleRKdpcMiVa86+0Zy8L8Z0tvs4Hts906fCeH0VFsqFl2nLSxwkHXgdCigLphkU1aOSeNwdMdp0n0+ilPf99yjeTteBmtDMQBzEIgpnjHQIX4luii60DgT4IEHLOaZ7Rqq77Q7RqhfecwAgZZvhJUi1/wAQ6fukkB11PYtd+NSoB4lxV+z7tUhi4ucenoqdfelg9xpd34LMr7z1TlDe4SepWdMdo2n7UsVEloAkGDABxGeJUm74Wfg7/r9VwFos4cSQTJx6qo+yP71Dc19G8Vif+j0//wAsonKeiDX3sbBusJOkkATplovLnhwzkKItLhkT1U+tXcS1hT+MjqbTXLiXOdJOJKw7Xabx8gqzbW4iCcEBxxnUZcllmzrJS6RtiwOFvtjVWuON4j2ruB788MsM01MOxDjkQMPM4qReeOs6ZoFSZnP+D9U/UhJa0S8eSPIg4xPOMx4JnHPDLPLXxUshHOfEZJOxmRnn4Rl0SXpP6G3mQNzTJF3IScW5deSG4VDADR7QkYjEdcM1aDpJkQXNjTLKBwzSZWEtIu+zkDz1zXRHHBP4mPqTkvkCsdJ2JqNkTBk5dw74R6dek0loptfmZwkCMsfFCMvhsjMklvzxyxCi6zXXexEa8eBwWe9ZL6/iE09eOSmKcuhowzHIaeULZs9INaAP5RaNxggtk8YxOGpUJWXlZ/USp8HT4eLVttclTaEy0SYg+oWFWrmcz1K3N [... cut ...] OCcFODikMJdUHAJi5AqvTEGvnil2hQWkaFSxTAc2nvSTYcCkpGUL6eUFiK0JAFa5K+oXVFxSAIcVNtIqsJlHDyBmmBCs2JUaYSfiisppDHugIUIhpoTjoiwHcQNEzGnM5qTMO9TbigBgFO/yThqkGIArVbRJgZpMYTnij9kAlTCYh2U1MIwpeKZ+CdAmDLU4HFMEem4RoeaAsdrhlKiXJ3O5D5oZn64JiKlZ0kkpUKcnkFaNEck4ATb/AAVEgQnc4KEKN5ShhQUzlBqaDqmBKVFzwmvDKAUN1Pw5BABLgzhEAQgU4egRMgpKHbJJgZjUenknSWY0O5AJxSSTQMnTzS18EkkMEEGYVluSZJJFAqhxQaaSSQCVlmXgkkmIHRPtFWSkkmIjVQ9R3/JJJDGi/oVU1SSVCQNp9pWGhJJL7G+gjfdPemCSSGJAbMZcfFSfl0SSQhsk5DCZJNEk25qJKSSGNA0VoSSQA6FVTpIBFYlJJJAz/9k=

superjax, 2017-02-21:
Precisely, but in many more pieces

superjax, 2017-02-22:
/home/ultron/Downloads/IMG_20170221_144414635.resized.jpg

superjax, 2017-02-27:
We have re-done all the wiring, and hopefully have gotten rid of the power issues.  (I'm pretty suspicious that it was an intermittent connection, because it seemed to have no correlation with distance to the transmitter, distance to the ubiquity wifi)  It is impossible to say exactly, especially with new wiring, but I think we have solved the problem.  We will perform 3 full battery drains on the ground while running the autopilot just to make sure this doesn't happen again.

closed 2017-02-27 by superjax