rosflight/rosflight_firmware #109: Sonar check reports found sonar when not connected

dpkoch, 2017-03-07:
I was playing around with the latest on branch `board`, and one time shortly after rebooting the Naze it reported finding the sonar when none was attached. Might be something goofy going on with how we check that? (@superjax what's the reason we don't just use the `init` function again but instead try reading and checking if the value is above some threshold?)

superjax, 2017-03-07:
I just fixed this today.

The reason we don't use the init function is because sometimes the sonar doesn't ACK.   The latest F1_only branch of breezy fixes the sonar present problem

closed 2017-03-08 by superjax

DaneSlattery, 2019-08-20:
Experienced this bug today

`rosrun rosflight rosflight_io _port:=/dev/ttyUSB0 _frame_id:=/base_link
[ INFO] [1566297275.507925779]: Connecting to serial port "/dev/ttyUSB0", at 921600 baud
[ INFO] [1566297275.529541460]: Firmware version: v1.1.0-40-gc20a35a0
[ WARN] [1566297275.537347972]: RC override active
[ERROR] [1566297275.537485108]: Autopilot ERROR: Invalid mixer
[ERROR] [1566297275.537554066]: Autopilot ERROR: RC lost
[ WARN] [1566297275.537660785]: Autopilot now in ANGLE mode
[ INFO] [1566297275.613421790]: Detected time offset of 1566296392.300 s.
[ INFO] [1566297276.303462035]: Got HEARTBEAT, connected.
[ INFO] [1566297276.512180776]: Received all parameters
[ INFO] [1566297384.522119341]: [Autopilot]: FOUND SONAR
`
Running on afroflight32 rev 6 (naze32)
and my companion computer is a raspberry pi 3b with ROS Kinetic.

This is on a copter, with no sonar connected at all. Does this have a potential to do harm?