rosflight/rosflight_firmware #110: ROSFlight as pure C++ Library

sytelus, 2017-03-08:
ROSFlight looks like a perfect firmware to integrate with AirSIm. Unfortunately, the current code is tightly coupled with STM32 and MavLink specific stuff. 

So I thought it would be great if it was possible to have firmware as pure C++ library. This library would just take sensor data and desired state as inputs and outputs control signals. All inputs and outputs are plain C++ data structures so it would have no dependency. One might use this library with Matlab code and someone else might use it for realistic visual simulation. 

This clear isolation of functionality would also allow fast debugging of core code without having to constantly deploy and test on actual hardware. We can also produce extensive unit test cases and enable continuous integration mode trivially. Supporting new hardware, sensors and actuators would become much more easier as well.

So I took the stab and converted ROSFlight in to pure C++ library: https://github.com/sytelus/ROSFlightCore. I've kept all functionality same so all changes are just geared towards isolating STM and MavLink stuff behind interfaces.

Here are the major list of changes:

1. All of the board specific code is isolated now behind pure abstract C++ interface called Board.
2. All of the MavLink specific code is isolated now behind pure abstract C++ interface called CommLink. This is important so we can support other (potentially better!) formats like Fast RTPS, MSPPG etc. Also this makes flight control bloat-free due to various MavLink ugliness.
3. Both Board and CommLink class are instantiable so that one can have multiple vehicles in simulator at same time. This is one of the highly desired feature.
4. Entire code base is now C++ with classes and without extensive use of externs and static variables. While doing this conversion, I uncovered few dependency and initialization issues that were buried in. Moving forward, I think this lightweight object model will help to better encapsulate private data used by module and clearly uncover dependency bugs.
5. I also resolved all warnings so code now compiles on GCC or VC++ in Linux or Windows and conforms to Level-4 warnings-as-errors standards.
6. Added new firmware class to encapsulate setup() and loop() functionality. This way, we don't have hardcoded while() loop in main() and other models would be possible.
7. All code is further converted to header-only library. This eliminates any need for builds and makes library drag-and-drop to any C++ projects. This is especially important in environments like Unreal where build systems are fairly complex.

I understand this is significant changes... Initially I tried to kept everything incremental but I realized a significant leap is needed so we can finally have a pure C++ cross-platform dependency-free firmware code that can be used in any project by simply copying header files (no builds required!).

Please take a look and let me know what you think: https://github.com/sytelus/ROSFlightCore. I would be happy to contribute back these changes in form of PR and delete my version so we have single master copy.

superjax, 2017-03-08:
Hey @sytelus,

This is awesome!  I think it's funny that @dpkoch and I both were working on a C-based abstraction layer while you were working on this.

I really like the idea of moving to a C++ library.  I just got finished navigating the sometimes gross linking hurdles required to get this C library working properly on our ROS/Gazebo simulator `fcu_sim`.   It's probably time that we get with the program and move to C++.

Probably my primary concern is that I need to see if this new library will still fit on the limited flash available on our little STM32F1 processors we use as flight controllers (I know, we should totally move to an F4-based controller, but we haven't had time to build up the driver library yet).  This is going to take a lot of work, but I'm willing to make it happen, both because putting our stuff in AirSim brings ROSflight more visibility and because as we are bringing up new hardware and supporting our own simulator moving to a more abstracted setup would be beneficial in the long term.

This also represents a pretty major shift in the way that we package the firmware, and we have to nail that down too, but again, I think it will be worth it.

I'm interested in what @dpkoch's thought are, but I'll try to incorporate your changes into a branch on the main ROSflight repo and see if I can't get a build going on actual hardware.  Is there any chance that you could be of assistance in this endeavor?

Thanks again, I'm excited to see if we can get this working.

superjax, 2017-03-08:
I suppose another thing is that ultimately, we would love to host the library and keep it connected to ours, so we can push updates to both the simulation and hardware simultaneously.

superjax, 2017-03-08:
I've started the `CXX` branch where I plan on working on this.  So far, I've switched everything above the board abstraction layer to c++, it builds, flashes and runs.  I didn't change anything to classes yet, but that will probably come this evening.

sytelus, 2017-03-08:
I would certainly like to help move to this new design. This would be awesome! ROSFlight could be the first firmware that is cross-platform, available as library and has nice abstraction that allows it to plug it in to any other programs like simulators or easily add new boards. 

I can take on all the work needed to make ROSFlight work with AirSim. This can allow us to do quick testing as firmware evolves. In the above code, I haven't implemented Board and CommLink interfaces for STM32 and MavLink respectively but that should be fairly easy. 

Yes, I agree that ROSFlight should be the place to host the library. I haven't publicized my repo precisely for that reason :). After we get to some stable state, I'll get rid of above repo to avoid any confusion.

superjax, 2017-03-10:
I'm about 60% done with the port to C++.  I haven't finished, but all the classes are built.  (I haven't linked up everything completely, you'll see a lot of commented-out code).  However, the framework is likely about finished.  There are certainly bugs in it still, but I think you could at least start getting the build system set up with the new C++ library.

superjax, 2017-03-10:
I have now finished (for the most part) the initial port to c++.  There are a few more things I need to work on in the mavlink files, but the main firmware is complete.  It will be a while before I can really vet this out on hardware.

I'm not a huge fan of the circular dependencies that are happening in the current architecture.  It would definitely be a good idea to think about the system architecturally and think of how these dependencies should really happen.

I also am doing some things I know I shouldn't.  (Like not using getters and setters for class variables).  Now that it's building, though,  that sort of cleanup should be easy.

superjax, 2017-05-18:
Sorry this has taken me a while to get to.  I have finally finished the C++ migration. 

We wanted to finish the 1st release candidate before pushing on the C++ conversion, because we didn't want to maintain two branches of the code.  Now that we have finished with version 0.1, getting this up and going is my top priority for ROSflight.

superjax, 2017-11-28:
This is now complete!

closed 2017-11-28 by superjax