rosflight/rosflight_firmware #112: Board revision

sytelus, 2017-03-09:
This doesn't look right...

In sensors.c

```
            params->set_param_int(Params::PARAM_BOARD_REVISION, (params->get_param_int(Params::PARAM_BOARD_REVISION) >= 4) ? 2 : 5);
```

Default revision is set 2 so this will make parameter oscillate between 2 and 5. 

May be it should be,

```
            params->set_param_int(Params::PARAM_BOARD_REVISION, (params->get_param_int(Params::PARAM_BOARD_REVISION) >= 4) ? 5 : 2);
```

superjax, 2017-03-09:
This is actually on purpose.  It's weird though, you're right.  I'm not sure exactly how to clean this up.

The IMU is connected to different pins depending on the board revision.  They switched it on revision 4, so 4 and greater are the same.  The intention is to change board revision, which would change where to look for IMU.

This definitely belongs underneath the board revision layer.

superjax, 2017-03-17:
This was moved under the abstraction layer in #103.  Changes will make their way to the `CXX` branch hopefully soon.

superjax, 2017-05-18:
Fixed

closed 2017-05-18 by superjax