rosflight/rosflight_firmware #116: airspeed measurement gets stuck

wynn4, 2017-03-16:
I was just looking over the rosbag from the most recent flight and noticed that the airspeed measurement kept getting stuck.  It would update a bunch (as it should), get stuck for some length of time (varying), update a bunch, and so on...

Please see the 'flat spots' in the following plot:

[image]

superjax, 2017-03-18:
I haven't been able to replicate this on the ground yet.  I'm wondering if it was a bad wiring connection.  (That would cause the same behavior.)  Perhaps I need to add a watchdog to the driver that reports zeros if the I2C fails to read.

superjax, 2017-03-18:
We might also consider hot gluing a molex connector to the naze so we can avoid intermiitent connections.

wynn4, 2017-03-18:
Sounds plausible.  Before the next test I'll wok with those guys to make sure we have a solid connection

superjax, 2017-03-23:
I just pulled out the airspeed sensor I've been bench testing with, and it seems to be working great.

I'm still suspicious that it was a intermittent connection, because I just ran this sensor for about 20 minutes without it freezing up like that.  If it's not the connection, then perhaps the sensor itself needs to be replaced.

At the very least, I am unable to replicate it on the ground with my hardware.

[image]

closed 2017-03-23 by superjax