rosflight/rosflight_firmware #117: Remove altitude hold mode?

dpkoch, 2017-03-17:
@superjax As I've been working on cleaning up the mux stuff, I'm starting to think we should remove the altitude hold stuff for now. I think it would be a nice feature, but (a) it's already bordering on "feature creep" anyway, (b) it really makes things a lot messier, and (c) the fact that we still haven't implemented it makes me worry it would hold up our 1.0 release too much if we're still going to promise it being in there.

So, what if we remove it entirely from the 1.0 release, then we can decide later if we want to put it back in? We can always get altitude control in the meantime by writing a controller on the ROS side that sends only throttle commands, which may just be a better long-term solution anyway. Let me know what you think.

superjax, 2017-03-17:
I would completely support that decision.  I wasn't entirely sold on the the idea to begin with, and my attempts at getting a Lyuapunov-stable altitude observer have not proven successful.

superjax, 2017-03-17:
We removed this as part of #103

closed 2017-03-17 by superjax