rosflight/rosflight_firmware #122: Documentation

superjax, 2017-03-19:
Hey @dpkoch, @el3ment, @dallinbriggs, @plusk01, @jerelbn. Thanks for interest in this project.  I am getting excited about an official release!  

As I've been thinking about RC1.0 (and helping people get going in the lab on ROSflight),  I am wondering if we should also release some real documentation.  My primary concern is the "user documentation" (rosflight.org, and the github wiki), not so much the doxygen "code documentation," which is more straight-forward, and less useful to newbies than a user guide.

I'm thinking the user guide (in whatever form it finally takes) could have the following outline.  I expect our  main audience to be people like the USU grad students, people in our lab, or people like Insitu, or AFRL who want to use our stuff.  I've got an example of what I'm thinking below.

My question to you is: 
1. Is this too much?  Too little?  Is there something I'm forgetting or don't need?
2. Would you be willing to help generate some of this?
3. Do you have ideas of how to present the documentation?  [ROSflight.org](http://www.rosflight.org) is really not that great right now.  It's super disorganized.  I suspect that there is a better tool for user guides online.

Anyway.  Thanks!

- Introduction
    + Purpose
    + Parts List
    + Best Practices
- Installing and running Software
    + Installing software (ros, eigen, 32-bit dependencies, stm32flash etc...)
    + Description of `fcu_*` and `ros_*` packages
    + Flashing ROSflight
    + Running fcu_io
    + Base Station Setup? (This is mostly ROS networking, maybe we can punt to the ROS tutorials?)
- Putting the Hardware Together
    + Wiring Diagram
    + Motor Layouts
    + External Sensors (GPS, Sonar, Airspeed)
- Preflight Checks
    + Common Parameters to check
    + Checking control surface deflections/motor directions
    + Perfoming Sensor Calibration
- Getting good performance
    + Multirotor Gain Tuning guide
    + RC Calibration and trim calculation
- Autonomous Flight
    + Provide control from onboard computer
    + Fly waypoints with `ros_copter`
    + Fly waypoints with `ros_plane`

- Advanced - These may or may not be important to RC1.0
    + Custom Mixers
    + Temperature Calibration
    + Saving parameter files and sharing them to other aircraft
    + Extending `ros_copter` and `ros_plane` example classes
    + Simulating ROSflight with `fcu_sim`
    + Developing custom firmware code
    + Extending ROSflight to other flight controllers (board layer API)

- Technical Papers (Appendix of sorts, these are latex documents that have already been written)
    + Estimator Paper
    + State Machine Paper

This is a rough outline, but I think that it would be sufficient to get someone up and going.  It might also be too aggressive.

jerelbn, 2017-03-20:
I think this is a great start. I'll help where I can.

dpkoch, 2017-03-20:
@superjax Yeah improving the documentation would be great. I'm happy to help. I think that outline is a pretty good starting point. One thing we might want to add is a "Getting Started" page that might just be a bunch of links to the other pages, but that provides a cohesive overview of the steps required to get something flying and talking to ROS.

@plusk01 Readthedocs could be good. Another good option might be mkdocs.org; with their system we could have simple markdown documentation that gets put into the readthedocs style, but we could host it wherever we want

superjax, 2017-03-21:
@dpkoch, I like the idea of a Getting Started page.  I added it to the outline above.

I also think that either of those options sound good.  Mostly, I really like the idea of using markdown.  @dpkoch, do you have time to get the readthedocs or mkdocs going?  The rest of us can work on generating the actual documentation.

Because it seems that we are all leaning towards markdown, I started an "Introduction" page, copied mostly from what is on ROSflight.org.  I have put it in this repo's wiki, [here](https://github.com/byu-magicc/ROSflight/wiki/Introduction).  (Turns out, you can clone and push the wiki as a repo, pretty cool eh?)   I would really love it if someone could take some time to review my Introduction page for me.  I'm not super confident in my ability as a writer, and would love a second opinion on how to make it better.

Otherwise, if anyone has a few minutes, and felt reasonably comfortable putting together a section of the guide, go ahead and grab a section and write it.  We can all review it and make it awesome, but I think a good starting point would be to just try and write it all down.  Just post here if you're going to grab a section.

Thanks!

dpkoch, 2017-03-21:
@superjax Yep I'll get that set up tonight. We could either host the documentation source code in a folder in the ROSflight repo, or as a separate repo. Do you have a preference?

dpkoch, 2017-03-23:
I've got us set up for readthedocs; see #126. I've also modified the outline some. Let me know what you think:

# ROSflight Documentation
Top-level bullets correspond to pages, second-level bullets correspond to section within a page.

## Introduction
- Introduction
  - Why ROSflight?
  - Objectives

## User Guide
- Getting Started
- Hardware setup
  - Parts list
  - Wiring diagram
  - Motor Layouts
  - External Sensors
- Autopilot setup
  - Compatible hardware
  - Flashing the firmware
- ROS setup
  - Installing/setting up ROS
  - Description of packages
  - Running fcu_io
- Configuration
  - Parameter interface
  - RC setup and calibration
  - Sensor calibration
- Preflight checks
  - Common parameters to check
  - Checking control surface deflections/motor directions
  - Calibration checklist
- Improving performance
  - Multirotor gain tuning
  - RC trim calculation
- Autonomous flight
  - Provide control from onboard computer
  - Fly waypoints with ros_copter
  - Fly waypoints with ros_plane
- Parameter descriptions

## Advanced
- Custom Mixers
- Temperature Calibration
- Saving parameter files and sharing them to other aircraft
- Extending ros_copter and ros_plane example classes
- Simulating ROSflight with fcu_sim
- Developing custom firmware code
- Extending ROSflight to other flight controllers (board layer API)
- Technical Papers
  - Estimator paper
  - State machine paper

dpkoch, 2017-03-23:
Also, in order to get a local live preview while you're working on it, you just need to `sudo apt-get install mkdocs`, then in the root folder of your ROSflight repo run `mkdocs serve` and open a browser to `localhost:8000` to see the preview. It will automatically update whenever you save changes to a file.

superjax, 2017-03-23:
That's awesome!  Thanks @dpkoch.

So, I'm thinking of grabbing 
* Hardware setup
* Autopilot setup
* Improving performance

Does anybody else want a topic?

superjax, 2017-05-07:
I just added latex support.  You have to install a python module before it will render properly.

`sudo install pip3 install https://github.com/mitya57/python-markdown-math/archive/master.zip`

Then, the normal mkdocs stuff should work.

dpkoch, 2017-06-14:
Closed by #126

closed 2017-06-14 by dpkoch