rosflight/rosflight_firmware #127: PARAM_MAX_COMMAND not converted to float

dpkoch, 2017-03-25:
It looks like PARAM_MAX_COMMAND got missed when everything got converted over to float; it's currently an int with default value of 1000. So it's probably doing nothing. I assume it should be a float with default value of 1.0? @superjax if you agree then I'm happy to make the switch.

superjax, 2017-03-25:
I don't think that MAX_COMMAND is used anywhere.  If that's the case, then we should just removed it.  (I guess then, what is the point of anti-windup?)

superjax, 2017-03-25:
Sorry, my mistake, it is used in the controller.  It should definitely be a float.   I think even a value of 1.0 won't do anything except in the most extreme cases, and I'm not sure what would happen to stability if we actually reached saturation limits....  Probably something I should test before release.

commit 65302cf5 (2017-04-07): fixed #127
PR #134 (2017-05-03): Saturation fix
Fixes #127
closed 2017-05-03 by dpkoch