rosflight/rosflight_firmware #128: Failsafe support

jerelbn, 2017-03-30:


plusk01, 2017-04-28:
It doesn't look like the `check_failsafe` logic in [mode.c](https://github.com/byu-magicc/ROSflight/blame/status/src/mode.c#L68) is right... 

```c
bool check_failsafe(void)
{
  if (pwm_lost())
  {
    _armed_state |= FAILSAFE;
  }

  else
  {
    for (int8_t i = 0; i<get_param_int(PARAM_RC_NUM_CHANNELS); i++)
    {
      if(pwm_read(i) < 900 || pwm_read(i) > 2100)
      {
        _armed_state |= FAILSAFE;
      }
    }

    _armed_state &= ~(FAILSAFE); // <-- This will always cause an out-of-range 
    return false;                // <-- signal failsafe to be ignored
  }

  if (_armed_state & FAILSAFE)
  {
    static uint8_t count = 0;
    if (count > 25)
    {
      led1_toggle();
      count = 0;
    }
    count++;
    return true;
  }
}
```

Also, it looks like we have three failsafe branches -- what is the state of everything?

superjax, 2017-04-28:
Thanks @plusk01.

You're right.  That logic is dumb.  Thanks for catching that.  We should update it on the `failsafe_support` branch and incorporate it into #131

The `failsafe_support` branch supports setting the hover throttle.  It should be ready to merge (besides this bug)

We are trying to decide how to handle #132, which would require offboard computer commands to be accepted even if we are in failsafe.  The work on that is on the `offboard_failsafe` branch and is still a work in progress.  Don't worry about that branch for now.

superjax, 2017-04-28:
@plusk01, do you any ideas on how to fix this?

plusk01, 2017-04-28:
How about

```c
bool check_failsafe(void)
{

  bool failsafe = false;

  if (pwm_lost())
  {
    // Set the FAILSAFE bit
    failsafe = true;
  }
  else
  {
    // go into failsafe if we get an invalid RC command for any channel
    for (int8_t i = 0; i<get_param_int(PARAM_RC_NUM_CHANNELS); i++)
    {
      if(pwm_read(i) < 900 || pwm_read(i) > 2100)
      {
        failsafe = true;
      }
    }
  }

  if (failsafe)
  {
    // blink LED to let the user know we are in failsafe
    static uint8_t count = 0;
    if (count > 25)
    {
      led1_toggle();
      count = 0;
    }
    count++;

    // Set the FAILSAFE bit
    _armed_state |= FAILSAFE;
  }
  else
  {
    // we got a valid RC measurement for all channels and pwm is active
    // Clear the FAILSAFE bit
    _armed_state &= ~(FAILSAFE);
  }

  return failsafe;
}
```

This builds. I can make a commit to `failsafe_support` if you'd like.

superjax, 2017-04-28:
This looks great!  Thanks, 

yeah, please go ahead and put a commit into `failsafe_support`

superjax, 2017-05-02:
Closed by #131

closed 2017-05-02 by superjax