rosflight/rosflight_firmware #132: Failsafe auto flight termination

mmmfarrell, 2017-03-31:
For fixed wing ideally we would like flight termination to be: throttle closed, full up elevator,
full right rudder, full right or left aileron. 

It would be nice to have different levels of fail safe, i.e. after 30 seconds of communications loss, the aircraft automatically Returns to Home or Returns To Land. After 3 minutes of communication loss, the aircraft terminates flight. 

Also would be ideal if RTH/RTL and flight termination were activatable by the safety pilot and ground station. 

(These are our requirements for AUVSI)

superjax, 2017-04-03:
I think the cleanest way to do this is to publish a `status` message from fcu_io.  This could include things like whether or not the safety pilot is in control, we are in failsafe, loop time, armed/disarmed, etc...  I think that the RTH/RTL and flight termination should probably occur within `ros_plane`.

The other change necessary is that we need to allow offboard control to be processed even if RC is lost.

@dpkoch, do you have any thoughts?

superjax, 2017-04-06:
I just added a status message which is published by `fcu_io`.  This includes a flag of whether or not the flight controller is in failsafe mode.  You'll have to watch this flag and act appropriately to fulfill your requirements.

As of right now, we default to RC.  However, we could allow onboard computer commands to be executed even in the absence of RC.

This should be supported by the `status` branches of `fcu_io` and `fcu_common` and the `status_message` branch of `ROSflight`.  That is, until we merge into `RC1.0`

superjax, 2017-04-07:
The latest `status_message` branch should allow for onboard commands to be accepted even if RC is lost.

mmmfarrell, 2017-04-07:
awesome. I will work on handling this message in ROSPlane

superjax, 2017-04-07:
When you get around to testing this, I can show you a quick hack that tricks ROSflight into thinking it's lost RC when you flip a switch.

superjax, 2017-10-03:
We should just make this a parameter so it can be turned on, but by default is turned off.

bsutherland333, 2023-10-04:
Failsafe behavior is something we'd like to revisit, so I'll leave this open.