rosflight/rosflight_firmware #148: Support Arbitrary IMU Orientation

superjax, 2017-05-03:
I had to mount my naze for my autogyro upside down.  How do I fly?

TimWhiting, 2017-11-22:
Wondering what the status of this issue is, because our team has the naze upside down on the foamy... Can I help get this working?

TimWhiting, 2017-11-22:
I looked into the branch for this fix. It looks like you were planning on supporting any arbitrary angle for the orientation. Is that necessary or could we just multiply the raw imu input by -1 based on an orientation parameter in the x y and z axis?

superjax, 2017-11-22:
Sort of. You have to multiply both the z and y axes by -1, (otherwise you'll end up with a left-handed coordinate frame).  Since there are so many configurations to consider, it just made sense to support arbitrary orientations.  There is actuallly a pull request active on this one, which adds 3 parameters.  If you merged `master` into that branch, it would probably work.

legoabram, 2017-11-22:
I had to make a small tweak, but I have it working now. The parameters say degrees, but everything is treated like radians. I figured degrees made more sense in this context, so I added a scalar. I've also made a few organization changes to match what I've been seeing elsewhere in the code. If you would allow it, how may I submit my changes?

superjax, 2017-11-23:
Why don't you submit a pull request and put me as a reviewer?

superjax, 2018-02-23:
Finished!

closed 2018-02-23 by superjax