rosflight/rosflight_firmware #153: Output scaling for servos on V-Wing aircraft

len0rd, 2017-05-05:
In ROSflight's mixer.mix_output, the calculated scale_factor is not applied to any of the servo outputs. This works for most aircraft cases, however would this become an issue on a v-wing type aircraft?
@superjax I cant remember who else you asked me to tag on this

superjax, 2017-05-05:
@dallinbriggs,

Do you have any insights on this?

superjax, 2017-05-06:
Also,  I think you're referring to a v-tail.  

For context,  The problem is that,  with a v-tail, it is impossible to apply both full rudder and elevator simultaneously.  Should we scale all the servo outputs like we do multirotors, or should we let 
them saturate?

dallinbriggs, 2017-05-11:
I think it should be scaled around the center point of the servo, not zero. So, if you're commanding full elevator and full rudder, one ruddervator will be fully deflected and the other one would be at the center of the servo position. From a side view of each control surface, the fully deflected ruddervator would look something like this: \---- while the other control surface would look something like this -----.

closed 2017-07-28 by superjax