rosflight/rosflight_firmware #161: Jitter between RC override active and computer control

srogers4, 2017-05-16:
The status message is constantly switching back and forth between RC override active and computer control while the computer control switch is in the active position. We increased the override deviation and saw the same result. We echoed rc_raw, and the channels are outputting constant values while the switching takes place. We are publishing off-board commands at 100 Hz.

srogers4, 2017-05-16:
We just set the rc transmitter trims to neutral (1500) and the problem seemed to disappear.

superjax, 2017-05-16:
That makes sense.  We enforce that the RC transmitter does not have trims while flying with offboard commands.  This is because otherwise the offboard commands don't know where the trim point is.

closed 2017-05-16 by superjax