rosflight/rosflight_firmware #164: No NVIDIA TX1 USB support

dallinbriggs, 2017-05-31:
Currently, I am finding that the Silicon Labs cp210x driver for the Naze32/Flip32 board is not supported by the kernel that is compiled for the TX1. It is best to use the UART on the Naze32 and connect that to an FTDI USB cable or go directly to UART1 on the TX1.

superjax, 2017-06-09:
Thanks for finding this @dallinbriggs.

This should be fixed with other boards that don't use the SI chip.  For example the F4 boards use USB directly, and don't go through a UART->USB conversion.  That should help in the future.

closed 2017-06-09 by superjax

dpkoch, 2017-06-09:
It'd be nice to figure out how to get the cp210x driver installed on the TX1 and include it in our documentation for other people who want to use this setup maybe? Or at least describe Dallin's fix in a troubleshooting section of our documentation