rosflight/rosflight_firmware #167: Arms on boot up if R/C transmitter is not on

wynn4, 2017-06-09:
When I power up the board and the R/C transmitter is switched off, the board Arms and stays armed until I switch on my transmitter.  I'm using a PPM receiver which has failsafes set such that when the receiver gets no signal from the transmitter, it defaults to outputting 'disarmed' on the arm channel.  Once I have turned on my R/C transmitter, but then switch it off again (which puts the receiver into failsafe mode) the board stays disarmed as expected.  Also if I arm the board with the transmitter on and then switch off the transmitter, the board disarms as expected.

superjax, 2017-06-14:
I can't replicate this on my flip32+ and an orange R618XL in PPM mode.  What receiver are you using, and what is the failsafe behavior set to?

dpkoch, 2017-06-14:
Slightly different, but might be related:

Using a CPPM receiver, if I power both the naze and the receiver at the same time (through the RC power rail), the red light comes on at boot, but the naze isn't armed. If I then set the `ARM_CHANNEL` param, the light goes off which suggest that `init_rc` might affect this somehow.

If I power the naze through USB, then power the receiver through the RC rail, the red light does not come on.

dpkoch, 2017-06-14:
Let's just verify this with @wynn4 and his setup before we actually close this; he was saying that his actually got armed. This may have fixed it, but let's just double check.

commit fb8f0b13 (2017-06-14): may have fixed #167
commit b377d782 (2017-06-28): Merge pull request #190 from rosflight/fix_167

Fix #167
closed 2017-06-28 by dpkoch