rosflight/rosflight_firmware #171: Check for calibration before arming

dpkoch, 2017-06-09:
I think we should check that an IMU calibration has been performed before the autopilot will arm. The easiest way to do this is to check if all of the calibration parameters are zero (their default value).

If someone tried to lift off with a multirotor without calibrating it wouldn't be too great; this would prevent that.

@superjax what do you think?

superjax, 2017-06-09:
It's actually not a huge deal in general, I fly manually all the time without calibrating, and just take care of it in stick trim.  You're right though, it'll mess up onboard computer commands.

I don't know. I don't want to make annoying to fly, adding additional steps could do that, but I'm not sure.

dpkoch, 2017-06-09:
Yeah, being annoying is a fair point. I'm not sure if it'd be too bad, since you just have to set it down level and then call the calibration service (or push the button in the ground station!). You'd also only really have to do it once after you flash new firmware, then the calibration would stick around forever unless you notice bad performance and decide to redo it. I could go either way though.

superjax, 2017-06-14:
That's right.  I like the idea that once you've at least done one calibration then it should work on boot up every time after that.

closed 2017-06-14 by superjax