rosflight/rosflight_firmware #191: Aux Commands

superjax, 2017-06-28:
It would be nice to be able to use the other PWM ports on the naze for auxiliary commands

(There is a prototype of this on the `aux_command` branch)

cmcquinn, 2017-11-17:
@superjax I've got a basic solution coded up on the `feature_191` branch. The idea is that you send a mavlink message with an array of floats representing the values for the PWM channels and an array of integers that indicate if each channel is a servo, motor, or neither. I wasn't sure how this fits in with the PWM ports controlled by the mixer, so if the channel is not designated as a motor or servo, then the function that handles the mavlink message does not touch that channel.

Is there any overlap between these 14 PWM channels and the 8 channels controlled by `Mixer::mix_output()`, or are they completely separate?

superjax, 2017-11-17:
There is overlap.  (the mixer controls the first of the 14 channels), but we actually want that.  I don't know if we want the aux command to override the mixer or not.  I think we do.  @dpkoch, do you have any thoughts?

superjax, 2017-11-20:
If you feel like the system is working, then go ahead and create a pull request, with `[WIP]` as the first word, and put me as a reviewer.

cmcquinn, 2017-11-30:
@superjax I merged `master` into my `feature_191` branch and got the new abstracted commlink layer. As far as I can tell, making this work with the abstracted commlink should be pretty similar to how I implemented it with mavlink, but do you have any input on how you want this done?

bsutherland333, 2023-10-04:
I believe this is already done.

closed 2023-10-04 by bsutherland333