rosflight/rosflight_firmware #194: IMU not responding error code 2

sethholsinger, 2017-06-28:
When trying to arm the motor using the RC transmitter I get the following error message: 'Unable to arm due to error code 2', This is the IMU not responding error code, but I can calibrate and receive data through the ROS node just fine. This could be a hardware issue as the naze32 I am using is quite old and I am not the first to use it, but I was wondering if you had any ideas as to what is causing this error? I should also note that I have the parameters setup to work with a fixed wing MAV.

superjax, 2017-06-28:
Do you happen to know the revision of your naze?

superjax, 2017-06-28:
So I think it was a problem with the way we were clearing that particular error on rev 5+ nazes. 

Do you think you could try my potential fix on the `fix_194` branch?

sethholsinger, 2017-06-28:
Rev 5. And yes I will try the potential fix.

sethholsinger, 2017-06-29:
Thanks for the possible fix but unfortunately it did not work. I get the same error when trying to arm.

superjax, 2017-06-29:
Okay.  Could you upload a copy of your param yaml file?  I'll see if I can replicate it.

sethholsinger, 2017-06-29:
I sent my param.yaml and parameters.yml files to you through email. Sorry I don't know how to do the same through github. Also I think I should note that I am using a virtual box and 14.04 with ros indigo, but I have also tested using on a non virtual box 16.04 running ros kinetic and got the same error.

superjax, 2017-06-29:
Cool.  (I think you can just attach it to a post).  I'll give it a try and see if I can figure out what's going on.

sethholsinger, 2017-06-29:
Here are the files. I just saw that my email didn't send properly. Sorry about that.  
[params.zip](https://github.com/rosflight/firmware/files/1113191/params.zip)

superjax, 2017-06-29:
Okay, I just tried it on my Rev5 naze in indigo, with the parameter set you sent.

After a clean build and flash of the firmware (`make clean`, `make flash`) and then loading the parameters `rosservice call /param_load_from_file` and calibrating the IMU `rosservice call /calibrate_imu` through the master version of `rosflight_io`  I was able to arm and disarm with channel 5.

Is there anything else that you are doing in your setup?

sethholsinger, 2017-06-29:
No, that is everything that I am doing for setup. I will try again using kinetic just to make sure, but I am now fairly certain that the naze32 I am using is damaged. I am receiving new naze32's either tomorrow or Monday and I will see if hardware damage is the issue. Thanks for all of your help.

sethholsinger, 2017-07-05:
I tried again with two new naze32 rev6s and I got the same error after flashing from the rosflight.hex files using the clean flight configurator, loading the parameters.yml file, and calibrating the IMU.