rosflight/rosflight_firmware #201: RC override automatically activating when autopilot is armed

sethholsinger, 2017-07-07:
When the autopilot is armed and commands are being sent to the /command topic the RC override seems to be on by default and the servos and motor will not respond to computer commands.  However, when the autopilot is not armed, computer control works and the RC override works when the yaw control is deviated. As noted in issue #198 by len0rd the RC override does not appear to be working properly when roll or pitch commands exceed the override threshold. Though the servos will still move based on the commanded RC inputs as well as the computer commanded inputs (i.e. if the ailerons are given an RC command they will respond to the RC command while the other servos respond to the computer control).

superjax, 2017-07-07:
Interesting.

Can you check your `/rc_raw`?   I wonder if one of your channels isn't centered.

superjax, 2017-07-07:
I'll look into this in the meantime.  This is new behavior

sethholsinger, 2017-07-07:
So I think I might have an explanation. The default threshold to initiate rc override is 0.1 deviation. If this happens on every rc channel and not just the four main channels that means that the arming channel passes over the threshold limit to trigger arming as the deviation must be 0.15 or greater (in my case as I am using a switch 1.0). Meaning that the act of arming is triggering the rc override. To fix this behavior I need to not use a switch as it stays over the override limit.

superjax, 2017-07-07:
yeah, you're right.  I think that is the intended behavior,  the sticks _are_ deviated, so it makes sense that you would be in override mode.

I assume that the rc override goes away after a little bit.  If that is the case, then is there a problem with the current behavior?

sethholsinger, 2017-07-10:
So I tried mapping the arming channel to a dial on my transmitter, and encountered some interesting behaviour. Basically in order to arm I need to put the dial to the max position and as soon as I move away from max deviation the system disarms. However, by being at max deviation on one of the rc channels (in this case 5) the rc override is going to be on constantly. Making so that while the system is armed it is impossible to have the system work on computer control. The intended behaviour does work when I map arming to one of the four main channels where when I deviate towards the positive direction it arms, when it goes back to 1500 it stays armed and when deviated negatively it disarms, but this make control of the aircraft more difficult as it sacrifices a control channel.

superjax, 2017-07-10:
Ah, okay.  Could you post a copy of /rc_raw before and after switching that channel? 

I have a hypothesis.  In fixing #167, I changed the values which are checked for arm and disarm.  The problem was that if the RC transmitter is not on, then the receiver outputs some values close to 1500, but not exactly.  This meant that sometimes the autopilot would arm without a transmitter connected.

Check out the change in PR [190](https://github.com/rosflight/firmware/commit/b377d78232c7f89bfb551a22256dd4fe8d9e7857)

I'll bet your transmitter is set to max out around 1850 and 1150 instead of 1000 and 2000.  Perhaps if we moved those values to 1750 and 1250 it would do better.

superjax, 2017-07-10:
Okay, would you try the updated `fix_194` branch?

(just do a `git pull` on the `fix_194` branch, it should update `rc.c` and do a reflash)

sethholsinger, 2017-07-10:
Okay. So I looking through mode.c in the firmware and I found out that arming only works on channel 3 (the rudder) by using the standard throttle down all the way push right all the way to arm push left all the way to disarm. Trying to arm and disarm through other channels cause the behaviours I have noted in my previous comments.

sethholsinger, 2017-07-10:
I tried the fix still seeing the same behaviour. Switch starts at 1502, I move it so that it reads 2015. At this point the rosflight_io node reads 'RC override active computer control lost Autopilot ARMED', I move the switch back so that it reads 1502 and the autopilot is disarmed and computer control is returned. If the move the switch to read 991 nothing happens.

superjax, 2017-07-10:
Hmmm...

Okay, I have two ideas.  Could you record a bag of you doing this?  I need /command, /rc_raw, /output_raw, and /status.  Also, send me a copy of your parameters again.

I'll look and see if I can determine what is going on from there, but otherwise, we can do another hangout.

sethholsinger, 2017-07-10:
Okay. Here is a rosbag of all of the data as well as the parameters that I am using. Let me know if you need anything else.
[issue #201.zip](https://github.com/rosflight/firmware/files/1136509/issue.201.zip)

superjax, 2017-07-11:
Okay, so I think I found the problem.

``` py
- {name: RC_ATT_OVRD_CHN, type: 6, value: 4}
- {name: RC_THR_OVRD_CHN, type: 6, value: 4}
- {name: ARM_CHANNEL, type: 6, value: 4}
```

The override channels and arm channel are both mapped to channel 4.  Try moving one of them to channel 5

sethholsinger, 2017-07-12:
It works now. Thanks!

closed 2017-07-12 by sethholsinger