rosflight/rosflight_firmware #203: C++ Unit Tests (Nice to Have)

superjax, 2017-07-11:
- [ ] IMU Calibration

- [ ] IMU not responding

- [ ] Time going backwards

- [ ]  Controller Anti-Windup

- [ ]  Equilibrium Torque Calculation

- [ ] Mixer Scaling

superjax, 2017-07-19:
I thought of some more unit tests

- [ ] barometer calibration

- [ ] barometer altitude calculation

- [ ] fastAlt function

- [ ] differential pressure calibration

- [ ] differential pressure velocity calculation

dpkoch, 2017-07-20:
One more:
 - [ ] Check important default parameters (invalid mixer, imu calibration parameters, etc.)

drinkdhmo, 2017-08-02:
- [ ] Initiate barometer calibration while calibration already running
- [ ] Initiate differential pressure calibration while calibration already running
- [ ] Initiate sonar calibration while calibration already running

superjax, 2017-08-11:
- [ ] Outlier rejection

superjax, 2017-08-20:
I just consolidated the checklist into the first comment.

bsutherland333, 2023-10-04:
This is something that should be completed at some point.