rosflight/rosflight_firmware #204: Motor assignment

dpkoch, 2017-07-11:
So, I'm thinking it might be nice to get rid of the weird baseflight/cleanflight motor mapping and replace it with the following (before our user base gets too big):

 - The front right motor is number 1, and motor numbers increase going clockwise
 - Odd motors are CW, even numbers are CCW

This matches our current Y6 motor mapping, for example. We can try to adjust this to match existing motor mappings as much as possible, but I think they're so mixed up right now that whatever we choose we're going to break half of the existing motor directions anyway. At least this way we'd be standardized going forward.

@superjax what are your thoughts?

superjax, 2017-07-11:
The only drawback is messing up people's current configurations.  I think that's okay.

When should this change take place?  Is this the first update following 1.0.0?

dpkoch, 2017-07-11:
Yeah, most likely? Or do we roll it into 1.0.0?

I'm also thinking we should switch the fixed-wing order to TAER. This matches the JR/Spektrum (and I think FrSky) default, and is less weird to me than AETR.

Also, we should probably also eventually support as a minimum:

  - quad x/+
  - hex x/+
  - octo x/+
  - Y6, X8
  - fixed-wing (standard configuration)

and maybe eventually add support for other configurations like V-tail or tri-copter, etc. But probably not those weirder ones at first.

superjax, 2017-07-19:
This is closed by #212

closed 2017-07-19 by superjax