rosflight/rosflight_firmware #210: Move sensor_present functions to sensors module

dpkoch, 2017-07-18:
In the board support layer right now there are both `sensor_check()` and `sensor_present()` functions, where the `sensor_present()` functions just returned the cached result from the `sensor_check()` function. The purpose for this is so you can call `sensor_check()` while disarmed, but call `sensor_present()` while armed so you don't hang the I2C bus while armed. Let's get rid of the `sensor_present()` functions in the board layer and move the caching to the `sensors` module to prevent incorrect implementations when someone creates a new board support layer.

superjax, 2017-07-19:
Also Move the calibration functions out of breezy, into the firmware

superjax, 2017-07-24:
closed by #213

closed 2017-07-24 by superjax