rosflight/rosflight_firmware #217: Crazy servo output when aircraft is disarmed

dpkoch, 2017-07-21:
Original post by @dallinbriggs:

I think something may be wrong with the servo output while sitting disarmed. Having a high pwm update rate causes the servo output to be completely random. In this plot, all I did was switch the update rate from the default frequency to 50 Hz for servos. 

[image]

dpkoch, 2017-07-21:
@dallinbriggs It appears that this behavior only occurs when the autopilot is in failsafe mode. This happens because the failsafe command is zero setpoints in angle mode, so the attitude controller is running. Can you confirm that this stops happening if you have RC on?

dpkoch, 2017-07-21:
@dallinbriggs Can you confirm that #218 fixes this issue?

superjax, 2017-07-24:
closed by #218

PR #218 (2017-07-24): Make separate fixedwing failsafe command
Closes #217
closed 2017-07-24 by superjax