rosflight/rosflight_firmware #225: Mixer numbers

dpkoch, 2017-07-31:
We need to finalize what the mixer numbers are before we do the next release, so that those can not change in the future.

Here's one proposal:

| # | Mixer |
|---|---------|
| 0 | ESC calibration |
| 1 | Quad + |
| 2 | Quad X |
| 3 | Hex + |
| 4 | Hex X |
| 5 | Octo + |
| 6 | Octo X |
| 7 | Y6 |
| 8 | X8 |
| 9 | Tricopter |
| 10 | Fixed wing (traditional AETR) |

The advantage of this is that all of the multirotors we'll ever support are taken care of, then it lets us gradually add more fixed-wing mixers (flying wing, v-tail, whatever) in the future while keeping a nice logical order without messing up existing assignments.

@superjax What are your thoughts? @dallinbriggs Do you have any additional insights?

superjax, 2017-07-31:
I guess it doesn't capture 10-rotor or 14-rotor monsters, tilt-rotors or others, but I think those will be weird enough that it doesn't matter too much.

dpkoch, 2017-07-31:
Yep! I think our policy is that if you want one of those, fork the firmware and do it yourself.

superjax, 2017-08-11:
Completed by #235

closed 2017-08-11 by superjax