rosflight/rosflight_firmware #229: Reboot To Bootloader

superjax, 2017-08-02:
This is becoming increasingly necessary as autopilots are buried in the UAVs

cmcquinn, 2017-08-03:
@superjax I added a service for this to the code in the rosflight repo. Can you add me to that repo so I can make a branch with my changes?

superjax, 2017-08-03:
Check your email

cmcquinn, 2017-08-04:
I successfully tested my fix on a board. The changes are in reboot_to_bootloader branch on the rosflight repo.

superjax, 2017-08-04:
Hey @cmcquinn, we typically don't "close" an issue until we merge the commit into a release branch (like `rc1.0`.  Thanks for putting this together.  When we get around to merging that in, then we will close this issue.

cmcquinn, 2017-08-18:
I just realized today that this doesn't work properly. Sometimes the code will flash without issues, but doesn't run properly, and other times `make flash` returns an error before the firmware can be written 100%. I am going to do some research and see what we can do to fix this.

closed 2023-10-04 by bsutherland333