rosflight/rosflight_firmware #243: Velocity and Altitude Control

superjax, 2017-08-11:
Add velocity and altitude control for multirotors

bsutherland333, 2023-10-16:
This is something that may be useful to have, but may also be outside of the scope of the project. The current goal is to provide the minimum functionality required for a safety pilot to fly the UAV through all portions of the flight and easily give/take control to an onboard computer. However, higher lever control like this may best be left to the software on the onboard computer, like ROScopter. More discussion on what exactly the "scope" of the project should be going forward will probably be required before we can decide whether something like this should be implemented.

bsutherland333, 2024-01-25:
Our current plan with ROSflight is to maintain the bare-bones nature of the firmware, and instead add this functionality in officially supported computer-side ROS stacks, in this case ROScopter. The goal of the firmware would then be just enough functionality for the safety pilot to be able to fly an aircraft through any portion of the flight, but not much more.

closed 2024-01-25 by bsutherland333