rosflight/rosflight_firmware #244: PixHawk hardware Port

superjax, 2017-08-11:


dagar, 2017-11-08:
I'm worked with PX4 and the pixhawk derivatives quite a bit. I've become interested in doing more with ROS. Let me know if you'd like some help here and we can discuss it in detail.

superjax, 2017-11-10:
Hey @dagar!

That is awesome!  I'd love to chat!  

I have quite a bit of progress trying to get the STM32F4 supported, and have been running into some linker errors.  I've been primarily testing on an Openpilot Revo (which has the STM32F405), while the pixhawk has the STM32F427 and the co-processor, and I haven't actually tried it at all on the pixhawk hardware.

Let's set up a time to talk,  feel free to PM me - superjax08@gmail.com

superjax, 2017-11-27:
An update on this issue.

@len0rd  and I have completed all the hardware drivers and unit-test-like examples for all the F4 hardware peripherals we need (SPI, I2C, PWM, UART, VCP and timers).  Trying to build it though, I am getting a super weird error.

This is my `main` function:

``` C++
int main(void)
{
  rosflight_firmware::Revo board;
  board.init_board();
  rosflight_firmware::ROSflight firmware(board);

//  firmware.init();

  while(1)
  {
//    firmware.run();
    board.led0_toggle();
    board.clock_delay(100);
  }
  return 0;
```

As you can see, I'm actually not running any part of the ROSflight core, I'm just creating it.  However, after the first line, the processor calls every single interrupt then finally hangs on the `HardFault` handler.

Relevant information:
 * I'm using the standard `stm32f405.ld` and `startup` files that are used in Breezy.   These linker files work for all my simple examples
 * my debugger (OpenOCD) is giving this error after exiting the first line `Error: address + size wrapped (0xffffffff, 0x00000004)`

Anyone have any ideas?

superjax, 2018-01-12:
Okay.  I have a simple example that isn't pretty, but I'm streaming IMU at 500 Hz no problem.

The parameters are pretty garbled up and I'm getting tons of errors because I've disabled stuff in the firmware, but it's at least reading IMU and printing it to the ROS network!

superjax, 2018-01-18:
This depends on #85

superjax, 2018-05-15:
I figured this out - It was a stack overflow due to some weirdness in the linker file.  For some reason, the linker file for STM32F4 supplied in 2015 puts the IRQ stack directly after the `text` section.  When the parameter struct is initialized, we'd go into the IRQ stack and immediately hardfault (from the comments in the linker, it appears that the ST engineers did this on purpose).

Just wanted to tie up that loose end.

bsutherland333, 2023-10-04:
Current plans are to port our H7 firmware to the Cubepilot Orange, once it's ready for open-sourcing.

bsutherland333, 2024-07-10:
This was implemented for mRo's Pixracer Pro board in pull request #424

closed 2024-07-10 by bsutherland333