rosflight/rosflight_firmware #249: Firmware update (remotely)

GabrielPaliari, 2017-08-17:
I intend to implement some code of several control theories modifying the Rosflight firmware code. The goal is to create a quadcopter (or more than one) that will be a research platform. I am a beginner at ROS and I have a Flip32. 
I want to know if its possible to **update the Firmware remotely**, without rebooting manually and through a command line in the onboard computer (Im using a rapberry Pi 3). 
This will probably save a lot of debugging time. 

Thank you for the help

superjax, 2017-08-17:
Hey @GabrielPaliari! welcome to ROSflight.  

I wouldn't say this was perfectly supported, but there _is_ a way to do it.  It's pretty clunky, but it works. 

The last pull request into master merged in #229, which added a `reboot_to_bootloader` service to `rosflight_io`.  If you call this, then the flight controller will enter bootloader mode, which can be used to flash the firmware.  Here are the steps

1. `rosrun rosflight rosflight_io`
2. `rosservice call /reboot_to_bootloader`
3. kill `rosflight_io` (to free the serial port)
4. `make flash` (in the firmware directory) or `   stm32flash -w rosflight.hex -v -g 0x0 -b 921600 /dev/ttyUSB0` with some custom `.hex` file.

I would imagine there is some way we could do these steps programmatically either with a standalone python node, or from within `rosflight_io`.  I don't have time to implement this right now, but I don't think it would be too hard.

GabrielPaliari, 2017-08-17:
Thank You very much for the help. I think this will be very useful and will make things easier :)

superjax, 2017-08-17:
You're welcome!

Good luck! I'd love to see your finished product, sounds like a cool project!

cmcquinn, 2017-08-18:
Hey @GabrielPaliari , I just realized this afternoon that the `reboot_to_bootloader` service does not actually work properly (see #229 ). Until we get this fixed, the only way to be flash code and be completely sure that it will work is by shorting the boot pins on your Flip32 before applying power to the board. Sorry for any inconvenience!

GabrielPaliari, 2017-08-19:
@cmcquinn thank You for the support. The Rosflight is already helping me a lot and after getting things done in the simplest way i will probably also try to figure out a way to Reboot the Flip32 from distance. Thank You for the atention!

bsutherland333, 2023-10-04:
Reboot to bootloader should be working for the H7 boards.

closed 2023-10-04 by bsutherland333