rosflight/rosflight_firmware #259: Barometer Cal Seems to Get Stuck

superjax, 2017-09-29:
So, it seems to me that if the barometer fails on the first try, it can never succeed.  Not sure why, but it seems to be a problem.

wynn4, 2017-10-12:
[image]

[image]
Also having this problem both with the flip32+ (with integrated baro) and with the flip32 (external baro).  It seems to happen on both, but more frequently with the external baro.  Below are some screen grabs of it happening on the flip32+ with onboard baro.  I didn't tell it to calibrate here, it just suddenly dove to a huge negative altitude.

[image]

Here I ran a rosservice call to calibrate the baro and it finally completed the calibration a minute later.  The whole time, the board was motionless so the "too much movement" thing wasn't due to the board actually moving or the pressure changing.

[image]

superjax, 2017-10-13:
Yeah, we need to find a way to replicate this on the ground.  I'll dig into the i2c driver and see if there's still some issues there.

wynn4, 2017-10-13:
Great. The problem seems to be fairly easily replicated in my experience. The plots above were both from a plain flip32 at my desk with nothing else connected to it.

superjax, 2017-10-13:
I just slowed down the barometer update a little bit, and am not leveraging the i2c queue as heavily.  Perhaps this will fix it.  @wynn4, can you confirm this fix on your hardware?  (`fix_259` branch)

superjax, 2017-10-19:
As an unrelated, but potentially useful fix, I just recently added support for the BMP280 barometer from the rev6 naze32.  This chip uses far less overhead on the i2c lines.  It might be be a little better about avoiding this issue.

superjax, 2017-10-24:
This was merged into master (illegally)

closed 2017-11-26 by superjax