rosflight/rosflight_firmware #266: IMU intermittently stops responding

BillThePlatypus, 2017-11-08:
Seen with a Naze 32, connected to an odroid.  The IMU stops responding for up to 2 seconds.  The airspeed sensor is connected; this may be related.

BillThePlatypus, 2017-11-08:
Tried replicating the issue with with no sensors attached, but it worked fine

BillThePlatypus, 2017-11-08:
Now trying with an airspeed sensor attached.  I am using rostopic to print the IMU and airspeed information, and so far there is no issue.  I'll leave it overnight.

BillThePlatypus, 2017-11-09:
No problem overnight.  Next I'll try with an RC receiver as well

superjax, 2017-11-09:
That is good news.  Let's try it now with the airspeed sensor connected to the plane, (just use some jumpers to extend it out to the new board and see if we can determine where the problem might be coming from)

BillThePlatypus, 2017-11-09:
This might be related.  I was testing the bug on the plane that originally showed it.  It was working fine for a while, but then the IMU completely stopped responding.  As I write this, I flipped the autopilot switch on the RC to get a timestamp for comparison, (4 minutes later) and that caused the IMU to start responding again,  then stop for .003 s, and then start again.

BillThePlatypus, 2017-11-09:
I just saw the recommendation to use the airspeed sensor from the plane on the new board.  I was trying to solder on headers for the RC connection and bridged  a 5V and ground pin, so I'll have to fix that before powering up the other board.

BillThePlatypus, 2017-11-09:
Also, it just lost RC for .02 seconds, and thinks that there is a sonar.

PR #269 (2017-11-21): update breezy to fix imu lockup with airspeed
Fixed #266
closed 2017-11-21 by superjax